Ultimaker/Cura #9835: Backend crash related to wall count and max deviation

degtyarev708, 2021-05-15:
### Application Version

Arachne_engine_beta

### Platform

Windows 10, RTX2070

### Printer

Ender 3

### Reproduction steps

1. Sliced mesh with a line count of 5 with default maximum deviation of 0.25
2. Sliced mesh with a line count of 4 with default maximum deviation of 0.25
3. Sliced mesh with a line count of 4 with edited maximum deviation of 0.1
4. Sliced mesh with a line count of 3 with edited maximum deviation of 0.1

### Actual results

1. Failed to slice, backend crash with return code 3221225477 and "The edge we're about to connect is already connected!"
2. Successfully sliced model
3. Failed to slice, backend crash with identical error
4. Successfully sliced model

### Expected results

Slicer should successfully produce the set wall count within the limits of the model. 

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

So far, this has only happened to me on a single model (stl included) and I havent been able to reproduce it on several other models Ive tried. Maximum resolution doesnt seem to have any affect, nor have other settings I've changed.
[Clamp backend error.zip](https://github.com/Ultimaker/Cura/files/6483772/Clamp.backend.error.zip)

fvrmr, 2021-05-17:
Hi @degtyarev708 thank you for your bug report.
I can reproduce this issue. I will discuss it with the team. 
Keep you posted!

Vandresc, 2021-05-18:
Devs see CURA-8244 for our internal reference

closed 2024-12-09 by github-actions[bot]