Ultimaker/Cura #9836: Bridge mode turns fan off instead of up.

Susanoo1337, 2021-05-15:
### Application Version

4.9.1

### Platform

Windows 10

### Printer

Ender 5 Pro

### Reproduction steps

1. Slice object
2. Set cooling to 25%
3. Enable bridge settings
4. Set bridge fan speed to 99%

### Actual results

Once the print gets to the layer with the bridge, the part cooling fan shuts off, it resumes at 25% speed once bridging is done. (same thing happens when bridge fan speed is set to 100%)

### Expected results

Fan should crank up to 99% and then back down to 25%.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Project File.zip](https://github.com/Ultimaker/Cura/files/6485673/Project.File.zip)

smartavionics, 2021-05-15:
Your 2nd and 3rd bridge skin fan speeds are set to zero.

[image]

Susanoo1337, 2021-05-15:
Silly me, thank you!

closed 2021-05-15 by Susanoo1337

Life0fBrian, 2022-03-04:
I do have the same issue but have fan for bridge being set to 100%.
[image]
However Cura 4.11 puts a M107 into the gcode having the fan to stop instead of running with 100% RPM.
Already thought that my fans died but then they turned on again.

smartavionics, 2022-03-04:
But what is the fan speed for the 2nd and 3rd bridge layers set to?

Life0fBrian, 2022-03-04:
Those settings are hidden and should not matter.
But I now played a bit with the multiple layer bridge settings and now it puts 100% fan RPM into the gcode and those M107 are gone.
That is kind of strange!
On first tests it treated that single bridge layer as a first layer and disabled the fan and a later bridge in the model was printed with 100% as intended.

smartavionics, 2022-03-04:
The fact that they are hidden doesn't guarantee that they have no effect.

Can you share the project file that illustrates the fan being turned off? (Save the .3mf file and zip it up to attach)

maruhe, 2023-02-28:
was there another update on this thread? i still got this error in Cura 5.2.1 and solved it by changing the bridge-fan-speed from 100 to 99
if needed I can upload a project as I can reproduce it again

walterp1982, 2023-03-12:
Confirming that for some time, when bridging fans are set to 100%, the fans are instead set to 0. I've been curious about this for months, and have just stopped using the bridging options. I wasn't aware that setting to 99% would rectify.
[V300_SmartTemperatureTower_PLA_180-225.zip](https://github.com/Ultimaker/Cura/files/10951473/V300_SmartTemperatureTower_PLA_180-225.zip)