Ultimaker/Cura #9837: Cura 4.9.0 and 4.9.1 crashes on exit

pc-bastler, 2021-05-15:
### Application Version

4.9.1

### Platform

macOS 10.15.7

### Printer

Creality Ender 3V2

### Reproduction steps

I copied the Program from the downloaded DMG to my programs folder, keeping my previous installation of Cura (4.8), so Cura 4.9.1 gets automatically renamed to "Ultimaker Cura **2**".

Then I start Cura 4.9.1 and Quit the program (it is not necessary to change any settings or work in the program to reproduce the error).

The same issue already happened with Cura 4.9.0 (4.8 works fine).

### Actual results

The program crashes with the attached error.
[error.txt](https://github.com/Ultimaker/Cura/files/6487927/error.txt)
[cura.log](https://github.com/Ultimaker/Cura/files/6487930/cura.log)

### Expected results

The program exits without any error.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

The code gets created without any issues when using 4.9.1, it is just the program exit that is faulty.

The logs are attached above at "Actual Results".

fvrmr, 2021-05-17:
Hi @pc-bastler thank you for your bug report.
I will discuss this with the team. Keep you posted.

pc-bastler, 2021-05-18:
I have additional Information in the meantime:

The issue happens when I work with my main user account (standard user), I upgraded the settings from 4.8 to the 4.9-branch.

The issue does NOT happen when I work as administrator on the same Mac. As administrator I did not had Cura configured previously, so 4.9.1 was a "First Run" for this privileged User. I set it the the same settings as I made for my standard user.

So the error could be caused by missing privileges or a problem with the migrated settings from 4.8.

Vandresc, 2021-05-18:
Deva see CURA-8245 for internal reference

drboone, 2021-05-20:
I've been seeing a fairly similar crash at exit on debian for a while.  4.8-4 installed via apt.  Log attached.

[curacrash.txt](https://github.com/Ultimaker/Cura/files/6517210/curacrash.txt)

MacPro-de, 2021-05-26:
Same issue here. Tested on macOS 10.15.7 (Intel) and on 11.4 (Intel and M1).
Strange is, that when use the Marketplace and click onto the "Quit Cura" button in the Marketplace window after enabling or disabling a plugin or after a successful installation of a plugin then Cura 4.9/4.9.1 quits properly without crashing (at least on all machines I tested). It seems, that only the use of "CMD-Q" and "Menubar > File > Quit" cause those crashes.

Ghostkeeper, 2021-05-31:
> I've been seeing a fairly similar crash at exit on debian for a while. 4.8-4 installed via apt. Log attached.

> curacrash.txt

This one has been fixed in Cura 4.9. Or rather, it should have been. The problem occurs in your version 4.8 because it mismatches the wrong Qt version with Cura, one which wasn't supported yet at the time.

ronbuist, 2021-06-05:
It doesn't look like renaming the image is causing the issue. I installed Cura 4.9 (and later 4.9.1) over a previous version (probably 4.8.x) and with me Cura also consistently crashes when I exit the application.

Errors and logging attached.

[error.txt](https://github.com/Ultimaker/Cura/files/6603047/error.txt)
[cura.log](https://github.com/Ultimaker/Cura/files/6603052/cura.log)

drboone, 2021-06-05:
Note that my case is not Mac-related.

@Ghostkeeper you're saying Debian improperly built the 4.8 release they're still shipping?

pc-bastler, 2021-06-07:
Meanwhile I swapped my Mac. With a "fresh install" I have the same problem, to summarize:

Now I am working on Big Sur (Intel Mac), latest patches to the OS applied.
Cura 4.9.1 newly downloaded and installed without prior installation of 4.8 or renaming the image, it still crashes on exit.

Ghostkeeper, 2021-06-07:
> @Ghostkeeper you're saying Debian improperly built the 4.8 release they're still shipping?

In a way, yes. Cura 4.8 needs Qt 5.10, not Qt 5.15. On Qt 5.14 or later it would crash on exit. [The Apt package](https://packages.ubuntu.com/hirsute/cura) says it needs PyQt >= 5.10, but that should be PyQt == 5.10, really. Of course they wouldn't do that, because that means that they would also need to somehow make sure that Qt 5.10 is still provided through Apt on those distributions, which is annoying for any maintainer. This sort of problem is one of the reasons why we went for AppImage instead.

It's all quite irrelevant to the issue at hand here which is Cura 4.9 and Qt 5.15. For some people it's working fine, but for some people it's crashing on exit. I'm wondering if it's only crashing on exit sometimes, or is it consistently crashing for these people every time?

pc-bastler, 2021-06-07:
>I'm wondering if it's only crashing on exit sometimes, or is it consistently crashing for these people every time?

For me it's crashing constantly on both machines (iMac 2013 with Catalina an MacBook 2013 with Big Sur).

ronbuist, 2021-06-07:
Same here: it’s crashing consistently on exit.

brucom62, 2021-06-11:
Was turning mad when found this ... same here with Catalina 10.15.7

MacPro-de, 2021-07-01:
I think, that I might have figured out, what is causing this:

I seems to be a localization issue.
Cura 4.9/4.9.1 and also 4.10 crashes on exit, when the language is set to "Deutsch" (German). This might also happen with other languages. It does not happen, when the language is set to "English".

If your Cura installation is set to "Deutsch" and you would like to set it to "English", then please first set the language in the Cura Settings. Then go to the Marketplace and enable or disable a Plugin of your choice. Then click onto "Quit Cura" in the Marketplace window. Only then Cura seems to properly save the settings and will use english language on the next launch.

After that it seems to quit properly.

ronbuist, 2021-07-05:
I can confirm that setting the language to English and closing Cura through the Marketplace has solved the issue for me as well, in both Cura 4.9.1 and 4.10. I had the language set to "Nederlands" (Dutch).

MacPro-de, 2021-07-05:
The problem seems to be caused by a mixup in the Cura Mac menu bar in (at least some of the) non-english versions.
When I launch Cura 4.10 in english language, then the Menu bar looks like this:

[image]

When I launch Cura 4.10 in german language, then the Menu bar looks like this:

[image]

The "Ultimaker Cura" Menu in english:

[image]

The "Ultimaker Cura" Menu in german:

[image]

The "File" Menu in english:

[image]

The "File" Menu ("Datei") in german:

[image]

Please note the last menu entry.
I think, that the menu entry "Beenden" should not appear at that position.
But, when I use the last menu entry "Beenden" (which is "Quit"), then Cura quits properly. It does not quit properly and crashes when I use the the menu entry "Beenden" in the "Ultimaker Cura" menu or use Command-Q.

When I launch Cura 4.10 in english language, then the "Preferences" (Cura settings) are located in the "Ultimaker Cura" menu.
When I launch Cura 4.10 in german language, then the Cura Settings are located in the menu "Konfiguration" which does not exist in the english version.

[image]

Ghostkeeper, 2021-07-08:
MacOS automatically moves all application menu entries that it recognises to its custom "Ultimaker Cura" entry. Cura adds the menu entries to the File menu, but MacOS says no, I'm gonna move it there instead. This is very annoying because MacOS seems to just trigger on several words like "settings", "preferences", "quit", etc. For instance, it moved the Settings Guide menu entry of my plug-in there too because it contains the word "settings". I had to replace the S with a Cyrillic character that happens to look the same to fix that.

I don't think that causes Cura to crash on exit though. The menu entry is only one of the ways to quit Cura. There's also the X button on the main window, and several buttons inside the interface itself such as after you've installed a new plug-in from the Marketplace. The application menu entry is just a button that triggers the same action.

Ghostkeeper, 2021-07-08:
It's been confirmed that the language setting causes this! Strange, but true!

MacPro-de, 2021-07-08:
> It's been confirmed that the language setting causes this! Strange, but true!

Thank you for confirming.

Here is what I experience:

Quit Cura 4.10 (set to Deutsch) by using...

Menubar > Ultimaker Cura > Quit Ultimaker Cura = crash
Keyboard shortcut "Command Q" = crash
Menubar > Datei > Beenden (in english = File > Quit) = quits properly
Closing the main window = quits properly
Using the "Quit Ultimaker Cura" button in the marketplace window = quits properly

closed 2021-07-23 by Vandresc

Ghostkeeper, 2021-07-23:
This issue has been fixed for version 4.11.

PR #10169 (2021-07-23): Fix crash on exit on MacOS if using Cura's translations
If you're using Cura in a different language, it was crashing on MacOS due to being moved to the special "Ultimaker Cura" menu that MacOS creates. These were the reproduction steps:

* Change Cura's language in its preferences menu.
* Restart Cura to complete the language change.
* Exit Cura. MacOS gives an error message then.

To fix this, certain menu entries should not be translated. Translating them causes Cura to update the menu entry's name upon closing down Cura, but the menu entry can't be found in its original place any more. The [... cut]