Ultimaker/Cura #9838: Minimum Support Area not reflected in print but is in Cura display

csterlin, 2021-05-15:
### Application Version

4.9.1

### Platform

Windows 10

### Printer

Creality Ender 3

### Reproduction steps

1.  Set Minimum Support Area to 5 mm2
2.  Found small area of support not showing up anymore on Cura display
3.  Printed. 

### Actual results

Found the support still being printed, and actually fell over ruining the printout

### Expected results

The small support area should not have printed

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Additional project files are at dropbox
https://www.dropbox.com/s/a4qhjzxw3rgt00n/Cura_Project_Files.zip?dl=0

Cura generated project file is at dropbox
https://www.dropbox.com/s/n5l8hop2d4vafce/CE3_Knight_PartA.zip?dl=0

fvrmr, 2021-05-17:
Hi @csterlin thank you for your bug report.
It looks like this is not support but your model.
[image]

csterlin, 2021-05-17:
Hmmm... You are right. It should have displayed as light blue. I missed
that. I apologize for cluttering your inbox. I will study any further
issues with better attention to details.
Thanks for your quick analysis and response.
Chuck Sterling

csterlin, 2021-05-17:
I meant to say that the red line would (not should) have displayed as light
blue had it been intended to be support.
Chuck Sterling

closed 2021-05-18 by csterlin

fvrmr, 2021-05-18:
Hi @csterlin no problem!