Ultimaker/Cura #9840: Cura 4.9.1 duplicating the same layer twice each time it raises the Z

karmakina, 2021-05-16:
### Application Version

4.9.1

### Platform

Windows 10

### Printer

Mingda Rock 3 Pro

### Reproduction steps

I sliced a model in 4.9.1, just a calibration cube. 

### Actual results

It put down the first layer (0.16), then on the second layer it failed to raise the Z and still read 0.16 for a second time while it ground out the next layer over the first one. Then it raised to 0.32, printed 1 layer, and failed to raise again (printing the 4th layer also at 0.32, grinding through the previous layers). It then raised up.

I sliced the same model in 4.9.0 and didn't have a problem.

### Expected results

It should have put one layer down at 0.16, then raised to 0.32 for the second layer, as 4.9.0 is doing. It also doesn't happen in any of my other slicers. 

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Gcode is attached. Project attached.

fvrmr, 2021-05-17:
Hi @karmakina thank you for your bug report.
I don't see your project file attached. Could you try adding it again.
Thank you in advance.

karmakina, 2021-05-21:
It says it cannot accept the filetype for gcode and .3mf

GregValiant, 2021-05-22:
Yes, you have to put them into a ZIP file and post that.