Ultimaker/Cura #9841: Slicing Issue and Brim placement

dangerousdee59, 2021-05-16:
### Application Version

4.9.1

### Platform

 Windows 10 Pro, RTX 2060

### Printer

Creality Ender 3 Pro

### Reproduction steps

1. Tried to multiply a file it worked
2. Rescaled Z for each model. (3 of them)
3.  40%, 50%, 60%
4. Sliced
5. Attempted to print. Brims were incorrect or missing
6. Checked CURA, Brims worked on unscaled models, did not work on multiple scales
7. A Model over printed on top of brim

### Actual results

A Model over printed on top of brim, misaligned .
[image]
[CE3_GuitarPickTrinityKnot.zip](https://github.com/Ultimaker/Cura/files/6488423/CE3_GuitarPickTrinityKnot.zip)

### Expected results

The brims and the model should have all been properly printed

[image]

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[image]

Asterchades, 2021-05-16:
A brim will only be generated for a part which has layers against the bed, and it would appear that your left-most piece has an empty first layer. The "Remove Empty First Layers" feature only works across the entire build plate, so won't apply any correction as the other two have extrusions present on the first layer.

Either the piece on the left is vertically offset from the bed, or the first layer is comprised of elements too narrow to be printed as a result of the scale. Both can be addressed by repositioning the model manually, either by setting the Z to 0 or even slightly below.

dangerousdee59, 2021-05-16:
Thank you. I will try that.

fvrmr, 2021-05-17:
I will close this issue, because of what @Asterchades suggested would probably work.
But feel free to reach out if you have questions.

closed 2021-05-17 by fvrmr

dangerousdee59, 2021-05-17:
Yes it works. Thanks