Ultimaker/Cura #9843: different result for single model and multi model

yawyd, 2021-05-16:
### Application Version

4.6.2

### Platform

Ubuntu 20.04

### Printer

Voron 2.4

### Reproduction steps

1. add model **A**
2. get result **1** for model **A**
3. add model **B**
4. get result **2** for model **A** and result **2** for model B

result 1: inner walls shown
result 2: inner wall disapper

### Actual results

model inner walls disapper if multi model added to cura

### Expected results

get same result (for each mdoel) no matter how many model added to cura

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Hi, I'm printing a plane model with inner walls in it.
everything goes well if I open one file with cura.
like below:
[image]

but inner walls disapper if another model adedd, like below:

[image]

here is project file:
[inner_walls_disapper_multi_model.zip](https://github.com/Ultimaker/Cura/files/6488720/inner_walls_disapper_multi_model.zip)

and log file:
[cura.log](https://github.com/Ultimaker/Cura/files/6488721/cura.log)

Thanks.

fvrmr, 2021-05-17:
Hi @yawyd thank you for your report. 
I can reproduce your issue. I will discuss it with the team. 
Keep you posted.

GregValiant, 2021-05-19:
"Not watertight and consists of 3 sub-meshes."  So I said to myself "No problem, I'll fix the model".  I exported it from Cura as an STL and started cogitatin'.
I tried 7 different programs to repair and slice and except for a single instance - all the gussets disappeared.  Prusaslicer, Cura 4.9.1, KISSlicer, MatterControl, IdeaMaker, along with MS 3D Builder and Netfabb (for repair), .
It was IdeaMaker that left the gussets in after a repair to the file.  It's still a mess, but the gussets (kind of) remained.
This is layer 18 of the "good" Cura file.  This was sliced without any changes.  Alternate Extra Wall is turned on and the extra wall is present on the odd layers.  There are areas that are printing over air.
[image]

Here is the PrusaSlicer version.  There was one "good" model on the build plate.  I didn't find a setting to alternate extra walls so they are every layer.  There are unsupported extrusions in this slice as well.  There are no gussets.
[image]

Here is the IdeaMaker version with one model on the build plate.  The green lines might be a second loop but only appear every 6th layer.  This would essentially print as a single wall model and the gussets look like they've been knit instead of extruded.  There is the added bonus of some air printing here too along with ugly gussets.
[image]

The model has problems and they appear to relate to the internal support gussets.  As near as I can tell, the aberration is the file that sliced with the gussets in place.  I think an attempt to print it would fail, but it does have gussets.

yawyd, 2021-05-20:
> "Not watertight and consists of 3 sub-meshes." So I said to myself "No problem, I'll fix the model". I exported it from Cura as an STL and started cogitatin'.
> I tried 7 different programs to repair [...]

Thanks for you research.
actually，many people include me printed these model out, here is the model link:
https://www.thingiverse.com/thing:3040294
the link above contains some gcode, curaprofile and Simplify3D factory file. you can take a look if you are interested.
what confused me most is, I can slice stl model and print it out pretty well.
I will try to "fix the mesh" if I can figure out what it is : ) (after some google)

Ghostkeeper, 2021-05-20:
This seems to be caused by the Merged Meshes Overlap setting. Setting it to 0 removes the issue.

The shapes on the inside (except the two big cavities) are all positive shapes, not holes, due to their normal vector orientation. So according to a computer they are not vacancies in the volume, but essentially "double filled", if that makes sense. A volume overlapping with another volume. Union Overlapping Volumes is disabled, so it should use the [even-odd rule](https://en.wikipedia.org/wiki/Even-odd_rule) for volumes, causing double-filled volumes to become open again. This is why it works with a single mesh.

However Merged Meshes Overlap looks for multiple meshes, and under certain conditions (multiple meshes with intersecting bounding boxes) tries to merge them together with some overlap. This merging does a polygon union too, which didn't take this fill rule into account.
The reason why it only happened with multiple meshes is because the Merged Meshes Overlap is omitted when there's just 1 mesh, for performance reasons.

closed 2021-05-20 by Ghostkeeper

yawyd, 2021-05-21:
> This seems to be caused by the Merged Meshes Overlap setting. Setting it to 0 removes the issue.
>
> The shapes on the inside (except the two big cavities) are all positive shapes, not holes, due to [...]

finaly got it, unexpectedly.