Ultimaker/Cura #9845: Let parameter search also find values instead of only the main item

MathijsG, 2021-05-16:
### Is your feature request related to a problem?

I'm always frustrated when I want to easily disable for example the 'skirt' again, and I can't find it with typing in 'skirt', only with typing the exact category which is "Buildplate adhesion".

### Describe the solution you'd like

I would like to see that the search bar in the print settings also searches through the actual values.

For some settings I can search for them already somehow, but many not. In my ideal solution it would also apply with numbers.

### Describe alternatives you've considered

An alternative would be a more accessible  print settings menu, that easily shows you the manual adjusted parameters. Now it's either the very basic settings, while you often need more, and the other extreme is having everything checked, so it get's cluttered.

### Affected users and/or printers

Every user that uses the search bar would benefit.

fvrmr, 2021-05-17:
Hi @MathijsG thank you for your feature request.
When I search for skirt and brim is selected it shows the buildplate adhesion. 

[image]

I think searching for parameters wouldn't be a good solution, because what do you do with settings that have a checkbox.

But I think adding a show all changed setting to the visibility settings could be an option. Could that be a solution for you?

MathijsG, 2021-05-17:
Regarding the checkbox, I think that's less of an issue as it's (at least in my usecase) more used for settings that have actual specifics in their parameters. For example if I read something online about removing the travel lines on my outer side, and I see being mentioned that the parameter you should pick is "No Skin", I would want to be able search for that, instead of the setting only showing up when I type in it's actual function name "Combing".

The show all changed settings would be nice as well, but it won't eliminate the full usecase for my current proposed feature.

How come the 'skirt' parameter DOES work already though (will check on my Cura installation as well)? While many other settings/functions can't be search for this way yet.

fvrmr, 2021-05-20:
I will discuss this with the team. Keep you posted!

fvrmr, 2021-05-25:
Hi @MathijsG I have discussed it with the team. 
And decided to defer this feature request. It is not a priority for us to add right now.

> How come the 'skirt' parameter DOES work already though (will check on my Cura installation as well)? While many other settings/functions can't be search for this way yet.

This works because if you search for text in a dropdown menu it works. So you have to search for the exact name. So searching for `no skin` won't work for combing but `not in skin` is working.

MathijsG, 2021-05-27:
Ah, that makes sense. Good to know that it already searches textual values. It used to be called "No skin", that's maybe why I was confused.

Ghostkeeper, 2021-05-31:
It searches through the setting name as well as its description in your current language. It doesn't search through historical versions of the names or other languages.

It should be noted that the Settings Guide changes the setting description (if displaying the guide in setting tooltips, which it is doing by default now). This also changes the search results.

closed 2023-07-13 by github-actions[bot]