Ultimaker/Cura #9847: Crash in case of authentication issues on the Ultimaker Account login

sanangeles84, 2021-05-16:
### Application Version

4.9.1

### Platform

Windows 10

### Printer

Not Important for this issue

### Reproduction steps

1. Suddenly I couldn't open Cura anymore, I got this error many times in the log 
`2021-05-16 17:40:06,932 - DEBUG - [MainThread] cura.OAuth2.AuthorizationHelpers.getAccessTokenUsingRefreshToken [61]: Refreshing the access token for [https://account.ultimaker.com]`
`2021-05-16 17:40:07,040 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService._storeAuthData [250]: Attempting to store the auth data for [https://account.ultimaker.com]`
`2021-05-16 17:40:07,042 - DEBUG - [MainThread] cura.OAuth2.AuthorizationHelpers.parseJWT [114]: Checking the access token for [https://account.ultimaker.com/check-token]`
`2021-05-16 17:40:07,151 - WARNING - [MainThread] cura.OAuth2.AuthorizationHelpers.parseJWT [123]: Could not retrieve token data from auth server: {"errors": [{"id": "016dbcd4-5d8f-4e48-b6b6-4f3dcdbc949d", "code": "tokenExpired", "http_status": "403", "title": "The given token has expired.", "meta": {"error_type": "AuthError", "url": "https://account.ultimaker.com/check-token"}}]}`
  until the software decided to crash after many attempts (323 if I'm not wrong)
`2021-05-16 17:40:07,546 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [66]: An uncaught error has occurred!`

2. In the cura.cfg I found this strange null on access and refresh tokens:
`ultimaker_auth_data = {"success": true, "token_type": "Bearer", "expires_in": 600, "scope": "account.user.read connect.cluster.read connect.cluster.write cura.printjob.read cura.printjob.write drive.backup.read drive.backup.write library.project.read library.project.write packages.download packages.rating.read packages.rating.write", "received_at": "2021-05-16 16:01:04", "access_token": null, "refresh_token": null}`

3. removing completely `ultimaker_auth_data` object from the config file solved the issue

### Actual results

Crashes, each time, until I modified the cfg like explained in "Reproduction steps"

### Expected results

Maybe a plan-b to open cura without Ultimaker Account in case of authentication issues or giving the chance to login again from scratch, instead of crashing after many attempt

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

The crash has been automatically reported when happend, I clicked on that option on the crash window, but in this case I'm reporting you what I understood and how I fixed it

fvrmr, 2021-05-17:
Hi @sanangeles84 thank you for your bug report.
Could you share your whole log file?

sanangeles84, 2021-05-17:
Sure @fvrmr 
[cura.log](https://github.com/Ultimaker/Cura/files/6496803/cura.log)

fvrmr, 2021-05-20:
Could you try to remove the 4.9 folder here: C:\Users\<your username>\AppData\Roaming\cura
That will reset your preferences.

Let me know if that works for you.

sanangeles84, 2021-05-20:
Actually as I said, I already solved my issue removing ultimaker_auth_data setting from cura.cfg, this made me use the app without having the account linked anymore, and I didn't lose all my other settings and configuration like you do deleting cura folder, so I was simply reporting the bug and the root cause.
So I don't need support but I was just suggesting developers what they should fix/improve on how handling better the situation when there is an error while trying to login to Ultimaker Account, giving possibility to start a new login process, or giving the user the possibility to start the app ignoring the account, instead of just crashing.

fvrmr, 2021-05-20:
Ah then I misunderstood, but I will discuss this issue with the team. So I will keep you updated!

jellespijker, 2021-05-20:
@sanangeles84 thanks for the feedback.

Fyi: the null value in your cura.cfg means that the actual value is stored in your system keyring.

Ghostkeeper, 2021-05-21:
Some reports say that it happens when you don't have an internet connection while starting.

fvrmr, 2021-05-21:
I created a ticket on our backlog to keep track of this issue. 
Devs see CURA-8255

closed 2021-05-25 by nallath

iceguru, 2021-05-30:
MY ticket got closed for this same issue and removing the appdata folder didnt help.