Ultimaker/Cura #9849: Disable USB Port Scan at startup

eboston, 2021-05-16:
### Is your feature request related to a problem?

When Cura starts up, I notice that some devices connected to USB get reset.  I'm guessing that Cura scans the USB port looking for a printer.  What ever it is doing, it is causing issues with devices it should not be affecting

### Describe the solution you'd like

Provide an option to skip USB port scanning when starting.

### Describe alternatives you've considered

There is no alternative I can think of.  Unplugging of the USB devices before starting Cura is no an acceptable alternative.

### Affected users and/or printers

Anyone with USB devices connected that are affected by the scan.

fieldOfView, 2021-05-16:
Go to Marketplace -> Installed. Scroll down and uncheck "USB Printing".

fvrmr, 2021-05-17:
Hi @eboston this is a duplicate of https://github.com/Ultimaker/Cura/issues/5207
And @fieldOfView comment is a workaround to solve this.

closed 2021-05-17 by fvrmr

eboston, 2021-05-17:
Obviously nothing has been done to truly address the problem.  How is anyone suppose to know to even look there?  What is so difficult in adding a item in Preferences, which is where most users would probably look for it, to turn it off?  And if UM is no longer supporting USB printing as was mentioned in #5207, why isn't this set to disable by default?

fvrmr, 2021-05-20:
> And if UM is no longer supporting USB printing as was mentioned in #5207, why isn't this set to disable by default?

I will discuss that with the team. Keep you posted.

fvrmr, 2021-05-21:
I have discussed it with the team. And the the reason why we this isn't disabled by default is because this is working for most people. More than for the people that it is not working for.

eboston, 2021-05-21:
While it is working for most people, they don't care if it is enabled or disabled by default.  But for the ones it doesn't work, who actually see a problem (first group doesn't see a problem), you leave them hanging.  There is no clear indication as to how to fix this issue.  A simple menu item in the preferences to disable USB printer would be so much easier to find than the current method.

Another way to thing of it.  As is, there is a large group with no issues and a small group seeing problems.  Disabling it by default would have everyone not seeing a problem.

fvrmr, 2021-05-21:
Changing it to disable by default, will cause issues for the people where it is working.

eboston, 2021-05-21:
Only if they are printing from cura.  Something that UM seems to be discouraging by removing USB printing from the printers.  Leave it enabled by default, but add a preference item to disable USB printing so that those who do have the problem can find the solution a lot easier than having to search the internet for the anser.

fieldOfView, 2021-05-21:
> but add a preference item to disable USB printing

You can already do that by disabling the plugin. But you have to go through the Marketplace.

One good addition would be for the USB Printing plugin to check if there are any printers configured that support USB printing in the first place.

eboston, 2021-05-21:
> > but add a preference item to disable USB printing
> 
> You can already do that by disabling the plugin. But you have to go through the Marketplace.

Which is the point I am trying to make.  It is not obvious.  As a user, I never added in a plugin so why would I think of looking in there?  A menu item in the preferences is much more intuitive.

> 
> One good addition would be for the USB Printing plugin to check if there are any printers configured that support USB printing in the first place.

For checking if there are USB printers, that could cause a problem.  If I have a UM printer defined (the plugin wouldn't be looking for non-UM printers I hope), but am not on the computer the printer is attached, then it could cause problems.  

Maybe a setting in the printer's machine setting would be a good location.  Add a menu item to indicate if USB printer is needed.  Maybe even have the user indicate the USB port or a button to scan for the printer.

fieldOfView, 2021-05-22:
> For checking if there are USB printers, that could cause a problem.

Printer definitions specify if they have USB ports. If only printers are added that specifically mention that they don't have a USB port (such as the Ultimaker printers since the UM3), then there is no reason to look for USB ports.

eboston, 2021-05-22:
> Printer definitions specify if they have USB ports. If only printers are added that specifically mention that they don't have a USB port (such as the Ultimaker printers since the UM3), then there is [...]

The situation where you have an UM printer that supports USB printing defined, but not attached, it will try scanning and could cause problems.

Ghostkeeper, 2021-05-31:
> While it is working for most people, they don't care if it is enabled or disabled by default.

No, this is likely wrong. Most people expect that Cura will print through USB cable out of the box. If it doesn't, then they can't print any more because they don't know that they should look through the list of plug-ins and enable a plug-in that gets shipped with Cura anyway.

It would be good to consolidate this discussion into #5207 , to put it in context with other proposed solutions.