Ultimaker/Cura #9851: Cutting/notching tool and USB printing optimizations

tacticalbacon89, 2021-05-16:
### Is your feature request related to a problem?

1.Yes and no its a problem in the way that it is a inconvenience that can easly be solved and boost Cura above other sofware 

2. Yes  cura by default as alot of issues reguarding printing from USB so much so that most user opt to avoid it entirely and buy rasberry pi setups  it is extreamly easy to interrupted a print while doing other thing on pc and also Cura client montering is taxing on the pc and will mostly cause you to just leave only Cura open and walk away  this would solve this and open door for future feature that would add value to Cura. 

### Describe the solution you'd like

1.can we please get the ability to cut parts on a plane into multiple stl parts  and add alignment notches so we dont need other 3rd party software

 i do understand that Cura is not a 3d modling software the cutting of stl's would not require a large amount of auto modeling code this type of implementation only need real modeling code in the case of adding notches which can be simplified to where it auto adds a default low poly joined notch design at the plane location selected to be cut and also auto applys a negitive to the other side before the final cut 

2.also the ablility/option  for cura to cache the gcode file in a different location when printing from USB  so that Cura does not need to be open and undisturbed while printing

this cache location could be ported from PC through USB and commanded to be written on to SD card that is inserted in printer via local motherboard then a command sent to launch print from SD card with matching file name with the only requirement for this to work being the need for a SD card inserted in printer.

basically to summarize you would be remote writing the SD card through USB  then sending a command to launch said writes remotely making the print process reliance local while launch process being remote

this would also open doors in the future to native WIFI writing support through Cura as the same process can take place but be sent through WIFI instead of USB.

 all you would need is to first send a command gcode packet coded to tell the firmware/motherboard to detect local SD card and boot a custom SD writing sequence using USB as port and wait on incoming file to be written, then when file is received through USB is then ported/written to SD card and then launched locally 

the result would be after the custom commands are sent, if i want i can shut my pc off or do something else like work on my next project ext. while my printer is printing. this would remove client monitoring for printing and  i will have full access to my PC without the need for a background program to continue running and be monitored using unnecessary resources and causing glitches in printing process based on what the client in doing or if its interrupted.

pros
local operation with remote launching
gcode part file would be stored on local SD to be available for later use if necessary

client reliance removal/glitches mitigated 
potential Cura native WIFI support (with motherboards with WIFI modules and read/write capability, no raspberry pi and octoprint needed)
potential for live rewriting of gcode file while printing via PC (being able to adjust printing process parameters or alter part mid print

cons
printer needs a SD card
slightly slower print start up
 lengthy coding 
small chance of file degradation

### Describe alternatives you've considered

1. Using other 3rd party software to cut and notch print 
2. Using raspberry pi with octoprint to remote launch prints

### Affected users and/or printers

All Cura supported printers and

fvrmr, 2021-05-17:
Hi @tacticalbacon89 thank you for your feature request.
For the next time please use one issue per feature request.
 1. is similar to this: https://hci.rwth-aachen.de/m3-autoparter
 2. You can use Octoprint for this, that would be a lot easier than what you are suggesting. 
Let me know if this makes sense to you.

tacticalbacon89, 2021-05-19:
1.That cutting addon is exactly what im talking about  but it should be
implemented into cura be default as a feature

2. Im aware that octoprint can accomplish  this but with a substantial cost
as purchases are required ie. Rasberry pi

Theres nothing difficult about coding this feature as the language already
exists all thats need is time for your development team to compile it in a
way that would make it work nativley this would make cura extreamly
powerful and would set it above its competitors

fieldOfView, 2021-05-19:
> Im aware that octoprint can accomplish  this but with a substantial cost as purchases are required ie. Rasberry pi

This is not true. You can run OctoPrint on any computer in your network, including the same computer you run Cura on, if you wanted.

nallath, 2021-05-19:
> 1.That cutting addon is exactly what im talking about but it should be implemented into cura be default as a feature 

Why should it be an integrated part? Doing so would mean that we have to maintain it and develop it further. 

> 2.  Theres nothing difficult about coding this feature as the language already exists all thats need is time for your development team to compile it in a way that would make it work nativley this would make cura extreamly powerful and would set it above its competitors

Just because all the bricks are there, doesn't mean it's easy or little work. Lets word it like that; Just because someone has a pile of bricks and wood, doesn't mean it's easy (or fast) to build a mansion from it.

tacticalbacon89, 2021-05-24:
**its longwinded, and full of grammar errors i know im sorry its hard for me to paraphrase a point im passionate about** 

1. because creating a all in one/ one stop software it what sets standards and creates extremely successful company from lazy ones
just look at Amazon as an example of that.  
and...
so what it take some small amount of extra time to maintain are the customers of your software not valuable enough for you to spend extra time giving them a extremely good quality product with features that set it above its competitors its called Craftmanship and cura has an entire development team to accomplish this task in a quick concise manner  

2. you are both right and wrong 
your correct in the sense that it doesn't make it require less talent you still need to know what your doing
but..
your wrong in the sense that it doesn't make it any faster or cleaner to do  

go build a house from complete scratch or conversely build a kit home you will find that the kit home can be built at almost 2x the rate and at reduced overheads *hints* why cookie cutter housing developments are so popular 

in the context of coding having the building blocks placed out in front of you and all you must do is compile them in working order using your experience as a guide  is a computer programmers dream come true, as it expedites the process and also allows for less debugging (so long as the building blocks are correctly coded themselves )

there are entire java, python and c sharp gaming development teams dedicated to this process of creating and utilizing building blocks because it is so successful   

3. i am aware that you can use octoprint on the pc via USB to accomplish this task, the point that i am trying to make is that by adopting that functionality to Cura means Cura becomes closer to a end all be all solution where all you need is Cura and a printer and you can do just about anything any other software can provide you and as popular octoprint is becoming/is if its developer decides to create a ''octoslicer'' Cura and most others will disappear if they even remotely do it well  

this would assist in making Cura the industry standard for printing and subsequently increase volumes of its software usage and Ultimaker printer purchases, as it lets the world know that you care about you customers experiences  and are not just a lazy team of devs trying to find quick money like the rest 

the point is ..
make yourself known as the only software you need to accomplish any challenge no matter how you wish to do it and
just because it requires some work doesn't mean you shouldn't do it it means another feature added to the swiss army knife of 
Cura

nallath, 2021-05-27:
>     1. because creating a all in one/ one stop software is what sets standards and separates extremely successful company from lazy ones
>        just look at Amazon as an example of that.
>        and...
>        so what it take some small amount of extra time to maintain, are the customers of your software not valuable enough for you to spend extra time giving them a extremely good quality product, with features that set it above its competitors, its called Craftmanship and Cura has an entire development team to accomplish this task in a quick concise manner.

I agree with your statement here, but i don't see how it supports either side of the argument. 

>     2. you are both right and wrong.
>        your correct in the sense that it doesn't make it require less talent you still need to know what your doing
>        but..
>        your wrong in the sense that it doesn't make it any faster or cleaner to do .

It does take more time. Source; My experience as one of the developers. 

> go build a house from complete scratch or conversely build a kit home you will find that the kit home can be built at almost 2x the rate and at reduced overheads _hints_ why cookie cutter housing [...]

I get that. But your vision of this is an oversimplification of reality. It's not "just" having to get it to compile (If only! My live would be _so_ much easier!) Every bit of code that you ship also adds extra burden of making sure that it _keeps_ working and also adds extra burden onto testing. That obviously doesn't mean that things can't be expanded, but it's not as cheap as you seem to think it is. 
 
>     1. i am aware that you can use octoprint on the pc via USB to accomplish this task, the point that i am trying to make is that by adopting that functionality to Cura means Cura becomes closer to a end all be all solution, where all you need is Cura and a printer, and you can do just about anything any other software can provide you, plus as popular as octoprint is becoming/is, if its developer decides to create a ''octoslicer'' Cura and most others will disappear if they even remotely do it well
> 
> 
> this would assist in making Cura the industry standard for printing and subsequently increase volumes of its software usage and Ultimaker printer purchases (im assuming you associated with Ultimaker printers as its in your name), as it lets the world know that you care about you customers experiences and are not just a lazy team of devs trying to find quick money like the rest

Sure, but _everyone_ uses that argument for a feature that they want ;) I think that we've already amply shown that we're not a team looking for some quick money. Hell; we are providing the SW for free to our direct competitors. 

Note that our own printers have integrated octoprint like solutions because it's just easier and more reliable. It also means that if we were to basicly re-implement this but then on the Cura side (which also has a number of downsides), that it wouldn't make things much better for us or our customers. I think it might actually be more expensive to build it in the way that you suggest than it would cost to send ever person that needs it a raspberri pi with octoprint pre-loaded. 

> the point is ..
> you should be searching for long exacerbated lists of features to boost Cura's functionality even if they seem worthless to you.
> as they may be useful in someone's task , make yourself known as the only software you need to accomplish any challenge no matter how you wish to do it and
> just because it requires some work, added cost or a challenge doesn't mean you shouldn't do it (thats called being a lazy dev or in some minds a cheapskate 'might of spelled that wrong'' ), it means another feature added to the swiss army knife of
> Cura

We can't support everyone. I don't like saying that, but it's the truth. We don't have the resources to give everyone what we want. As such, we have to make choices in what we do (and therefore; what we don't do). This means that we try to build things that we think have the most value for the most people. That is also why we have the plugin system, as it enables others to build additions onto things that we made (and perhaps also prove us wrong that a certain feature is worth it!)

So basicly, i disagree with the notion that this has to do with developers being lazy. You might want to rethink the way that you word this, as i hope you can see how saying that is uncalled for (and a bit insulting as well)

Ghostkeeper, 2021-05-31:
Duplicate of https://github.com/Ultimaker/Cura/issues/2472.

closed 2021-05-31 by Ghostkeeper