Ultimaker/Cura #9852: Saved projects with support blockers are useless / can´t slice

DaLimboster, 2021-05-16:
### Application Version

4.9.1

### Platform

Windows 10

### Printer

Ender 3

### Reproduction steps

Open a STL and add support blocker (i chose chaning infills). Save the Project, Exit Cura und Open the Project. The whole thing is one object and you cant slice it. Sorry for my englisch, i have german Cura.

### Actual results

Projects with Support blockers are useless.

### Expected results

Load a project with support blockers like you created it

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

no additional information

fvrmr, 2021-05-17:
Hi @DaLimboster thank you for your report.
Could you share your more info on what is happening with screenshots.
Maybe it is related to this issue: https://github.com/Ultimaker/Cura/issues/9767

closed 2021-05-31 by no-response[bot]

Ghostkeeper, 2021-05-31:
It would help to share the project in question, perhaps. It could be that you're accidentally exporting as 3MF (File -> Export) rather than actually saving a project file (File -> Save Project).