Ultimaker/Cura #9854: Setting 'Max Comb Distance w/o Retract' has no function

FPrefect20, 2021-05-16:
### Application Version

4.9.0

### Platform

Win 10

### Printer

n.a.

### Reproduction steps

1. Activating the setting "Max Comb Distance With No Retract" (entering e.g. "10mm")
2. Checking the preview

### Actual results

Ignoring the setting at "Max Comb Distance w/o Retract'" CURA is using 'Combing Mode' to travel all around the object.

In addition:
IMHO this calculated 'combing traveling path' is nonsensical.
The traveling path leads all around the object back and forth with no printing.

### Expected results

using the setting "Max Comb Distance With No Retract" like in other CURA versions

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]

FPrefect20, 2021-05-16:
project file:
[Adapter.zip](https://github.com/Ultimaker/Cura/files/6490231/Adapter.zip)

GregValiant, 2021-05-17:
Any horseshoe shaped object (like your split ring) can generate a lot of combing moves.  I generally turn it off for such models as I'd rather deal with a bit of stringing across the opening that all the extra motion.  Another reason to turn it off is that long combing moves (like you will have on that part) tend to allow some oozing and the prime move is not enough to bring the pressure back up in the nozzle resulting in possible under-extrusion when extrusion starts up again.
As for Max Combing Move without Retract, it does as it says.  If a combing move is beyond a certain distance (like your 10mm example) there will be a retraction and if it is under 10mm there won't be a retraction.  It has no effect on the movement, just whether a retraction should occur.
This shows the travel moves when your 3mf file is sliced.  All travel moves stay within the part.
[image]

And this shows the travel moves with combing turned off.  All travel moves jump the gap in the part.  According to Cura, doing it this was saves 4 minutes.
[image]

fvrmr, 2021-05-17:
Hi @FPrefect20 thank you for your report.
What @GregValiant says is correct. 

The `Max Comb Distance With No Retract` allows you to choose a distance of travel moves, after the travel move will be retracted. 

You can read more about this setting in the settings guide: https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/travel/retraction_combing_max_distance.md

Let me know if that makes sense to you.

Asterchades, 2021-05-17:
It seems to me like they're expecting the setting to impose a maximum distance for a combing travel, after which point the whole thing gets thrown out and it goes back to a standard retraction move. This is, of course, not what either the setting name nor its internal description state, much less what the Settings Guide or online documentation state it does, which is what they're seeing in that the combing travel is still taking place just now it's being retracted.

FPrefect20, 2021-05-17:
@Asterchades:
you are absolutely right! i was lost in translation using German language in CURA [4.9] / English in CURA [master].
The setting i was expecting in CURA [4.9] is this:

[image]

FPrefect20, 2021-05-17:
in case this feature didn`t made it to CURA [4.x] versions: 
- is there a way to stop the engine from travelling without need?
- is the unneeded movement (see project layer 13 for example) treated as a bug?
(IMHO this is a bug - even though with a different title)

Asterchades, 2021-05-18:
That setting is one which @smartavionics has added specifically to his builds, rather than part of the base Cura feature set. Unfortunately he no longer submits feature pulls to Cura (likely because of how much his builds differ from the original these days) so it would need to be added as a Feature Request by the main Cura team.

Ghostkeeper, 2021-06-01:
So if I understand the development of this thread, the issue is now a feature request to implement a "maximum detour distance" type setting, that chooses the straight travel move rather than the combing travel move if the combing travel move is significantly longer than the straight move?