Ultimaker/Cura #9857: Job Name

vincesmithmail, 2021-05-17:
### Is your feature request related to a problem?

I have many old G-Code files that I spent a lot of time to get right.  I can't easily tell what material they are set up for If I want to go back and print something again.

### Describe the solution you'd like

In general settings you have a tick box for "Add Machine Prefix to Job Name"  Something similar with basic material name like ABS, PETG, PLA, NYL, TPU....etc...

### Describe alternatives you've considered

I try to remember to do it myself when I am saving the file

### Affected users and/or printers

All....certainly far more users than those who need to have the machine prefix.  (There are far more users who have multiple materials than there are who have multiple printers)

### Additional information & file uploads

It may be something you have to set up in the manage materials section.  The Prefix could be one of the fields associated with a given material.

fieldOfView, 2021-05-17:
There are two plugins in the Marketplace that will let you do this automatically: Printjob Naming and Gcode Filename Format.

fvrmr, 2021-05-18:
Closing this issue since the plugins @fieldOfView suggested will solve your problem!

closed 2021-05-18 by fvrmr