Ultimaker/Cura #9858: Spiralized 1st Layer for seamless surface (like an analog vinyl record)

FPrefect20, 2021-05-17:
### Is your feature request related to a problem?

Seeking for the "Perfect 1st Layer"
If using "concentric" mode there is always a little line visible where the nozzle is moving to the next inner circle.
The feature "concentric-spiralized 1st layer" would eliminate this "line"
(within the resolution of the steppers)

[image]

### Describe the solution you'd like

Feature "concentric-spiralized 1st layer"
One single not-interrupted print from inner to outer circle (or reverse) like the analog vinyl records *) track 

*) if you really don´t have an idea of what i mean by "analog" or what a "vinyl record" might be: 
pls. ask somebody over 30 years  ;-)

### Describe alternatives you've considered

n.a.

### Affected users and/or printers

all users; all printers

fvrmr, 2021-05-18:
Hi @FPrefect20 thank you for your feature request.
But I have to defer this issue, because we don't think it is that important to implement this issue and not a priority for our backlog.
Hope you understand.

oliof, 2022-03-30:
Hi,
I would kindly request to revisit this feature request. My reasons are the following:

 * with the broad availability of CHT nozzles, printing with wider layer widths and higher layer heights is becoming increasingly popular
 * one of the drawbacks of CHT nozzles and similar designs is limited retraction performance
 * retraction is limited or not happening at all when switching between concentric perimeters
 * to have perimeter switchover line up you have to disable seam hiding (set it to off), otherwise some perimeters will start at a different location than the previous ended, which will lead to nozzle drag and due to the limited retraction performance cause unwanted filament deposition.
 * continuous spiralized extrusion with well tuned overlap will be especially useful for vasemode because there is a lower chance for holes, increasing the likelyhood for a waterproof print.
 
 And as an extra request, it woul be great if for vase mode spiral infill for multiple bottom layers would switch direction (outside-in and inside-out spiral) so that the last bottom layer before the spiralizing begins goes outside in.

toto-w, 2023-06-29:
Please don't remove this issue improvement. I would like to have it in a future release. It will improve printing quality and is really no rocket science.

Best regards
Torsten

gaaat98, 2024-02-22:
Hi! I am struggling with the same problem on all my prints with concentric first layers, and I would really love to see this implemented! Please consider doing it!