Ultimaker/Cura #9860: Missing 4.9.x Appimage Signing Key

brentalanmiller, 2021-05-17:
### Application Version

4.9.1

### Platform

Linux

### Printer

na

### Reproduction steps

I'm upgrading from 4.8.0 and downloading the new 4.9.1 AppImage from the github releases page. I cannot verify the signature because there's a new key used for signing (`3696EBEDD7C841E1`) and this key doesn't appear to be available on any of the common keyservers, nor could I find a reference to it in Google.

### Actual results

Can't verify signature.

### Expected results

Be able to download this key from the public keyservers and have some sort of web of trust associated with it. If its not possible to have a web of trust associated with it, having the public key on your web site would be good too.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

.

fvrmr, 2021-05-18:
Hi @brentalanmiller thank you for your report.
I will discuss it with the team. Keep you posted.

evtrados, 2021-05-20:
Hi @brentalanmiller,

Indeed, you're right! We changed build servers between the 4.8.0 and 4.9.0 releases and the old private key was not properly imported as it seems. I've imported it now and the signature should be what you expect now.
I've uploaded it on the 4.9.1 [release page](https://github.com/Ultimaker/Cura/releases/tag/4.9.1) as well.

brentalanmiller, 2021-05-20:
Awesome, thanks! I just downloaded the new signature and it shows a good signature with the old signing key `C1A1B91069C4AF59`.

closed 2021-05-20 by brentalanmiller