Ultimaker/Cura #9865: Magic Trackpad three finger drag broken

CNCXPROUK, 2021-05-18:
### Application Version

4.9.1

### Platform

Big Sur

### Printer

CR-10s

### Reproduction steps

Control and three finger drag no longer works. 

When I try and three finger drag a model to orientate it, Cura is really slow to move. It's unusable. 4.8 is totally fine. This happens in a new project or existing.

### Actual results

You can't rotate anything by three finger drag

### Expected results

Rotations should happen, they either don't or are extremely laggy

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

No project file needed

[cura.log](https://github.com/Ultimaker/Cura/files/6501774/cura.log)

fvrmr, 2021-05-18:
Hi @CNCXPROUK thank you for your report.
I will discuss this with the team. Keep you posted!

Ghostkeeper, 2021-06-01:
This was re-tested by one of our test engineers, with the following response:

> You need to enable the three finger drag option on the trackpad in order try and reproduce this. After I did that I could drag the model without clicking on the orientation arrows, but I cannot define the motion as slow. I experienced it as normal move speed and I also have one of the earliest MacBooks that are even allowed to update to Big Sur (early 2015), so I don't know why the OP experiences this. Maybe it is more evident in newer MabBooks?

We're asking someone with a newer MacBook in our team to try to reproduce the issue at the moment.

closed 2024-12-09 by github-actions[bot]