Ultimaker/Cura #9867: No profiles will work when I select eSUN PLA+ materials

CoilHeadd, 2021-05-19:
### Application Version

4.9.1

### Platform

Windows 10 

### Printer

Ender 3Pro

### Reproduction steps

I have started to use the materials where it sets the printer us for that specific brand and make of filament but for some reason when I select eSUN PLA+ it says not supported as you can see in the image so is there a way around it, please.

### Actual results

.

### Expected results

The expected result is being able to use my pre-made profiles for printing certain kinds of prints from Lithos and ect.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

[image]

fvrmr, 2021-05-20:
Hi @CoilHeadd thank you for your report.
This is because there are no profiles defined for the material. But not supported doesn't mean that you can't print with it. 
You can create a profile for this material yourself, but there a no pre made profiles for this. 
Here you can read more information about importing materials: https://github.com/Ultimaker/Cura/issues/8229

Let me know if this makes sense to you.

GL1Elwin, 2021-06-02:
from my experience; try duplicate the materials, and remove the '+' character from the name. it should work. it's a strange bug that persist since 4.8

Ghostkeeper, 2021-06-03:
If you duplicate a material, the material type is not changed so it should still have the same profiles as the original set.

If you edit the "Material Type" field, for instance to change from PLA+ to PLA, then you'll get the profiles intended for PLA instead of the profiles intended for PLA+. In this case there are no profiles for PLA+ so then indeed you'll get a set of PLA profiles rather than an empty set.

This is a duplicate of #8229.

closed 2021-06-03 by Ghostkeeper