Ultimaker/Cura #9868: Cura 4.9 - Ultimaker 2+ - Material profile will not save filament diameter value of 1.75mm (modified printer)

drphil3d, 2021-05-19:
### Application Version

4.9

### Platform

Windows 10

### Printer

Ultimaker 2+

### Reproduction steps

1. Duplicate Generic PLA Material Profile
2. Set the material diameter to 1.75
3. click ok on warning

### Actual results

Profile defaults back to 2.85mm

### Expected results

Save the value of 1.75mm after warning message

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I have an Ultimaker 2+ stock firmware with a with direct drive extruder for 1.75mm filament

fieldOfView, 2021-05-19:
> I have an Ultimaker 2+ stock firmware with a with direct drive extruder for 1.75mm filament

Have you set the filament diameter to 1.75 in the Extruder tab of the Machine Settings?

drphil3d, 2021-05-19:
I will try that, it hadn't even crossed my mind.

fvrmr, 2021-05-20:
Closing this issue, since @fieldOfView suggestion should work for you.
If you have more questions please reach out.

closed 2021-05-20 by fvrmr