Ultimaker/Cura #9869: Density

DjSteker, 2021-05-19:
### Application Version

4.9.1

### Platform

windows 10

### Printer

Prusa i3

### Reproduction steps

1. laminate 

### Actual results

density is low, leaves spaces between "walls" without filling, as if I used a 0.6mm nozzle when I use a 0.4mm nozzle.
I have to raise the flow to 300% when in version 4.8.0 of Cura it was 100% 

### Expected results

-

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

-

GregValiant, 2021-05-19:
In Cura, with the model loaded, use "File | Save Project" to make a 3mf file.  Then put the 3mf file into a Zip file and post it here.

DjSteker, 2021-05-19:
[Gcode_Cura_4_9_1.zip](https://github.com/Ultimaker/Cura/files/6511534/Gcode_Cura_4_9_1.zip)

GregValiant, 2021-05-20:
I also run Win 10 but I can't see a problem in that file (but my GPU is in compatibility mode).  Can you post a screenshot that shows the problem?  It looks like a solid mass of plastic to me.

fvrmr, 2021-05-20:
Can you give indeed a bit more information about what the problem is to you. Like @GregValiant mentioned screenshot would help.

DjSteker, 2021-05-21:
I have try and it seems to change some detail that makes the extruder not bite the filament well, the filament is flexible and slippery a lot

DjSteker, 2021-05-21:
It seems like it's fine, thank you and excuse me

fvrmr, 2021-05-21:
Good to know, then I will close this issue and mark it as solved!

closed 2021-05-21 by fvrmr