Ultimaker/Cura #9873: ADAPTIVE LAYER FEATURE

salvatore54, 2021-05-19:
### Is your feature request related to a problem?

>1 I have a few parts to 3d print,and some parts have thin horizontal layers,and I just need that part with a 0.1 mm height,but I don't have any setting in Adaptive Layers to set Which layers I want to do adaptive.

>2 And another problem in adaptive,I just have Maximum variation,but I don't have a minimum control,
because sometimes I want to print, all the layers with 0.2,and just a few parts with 0.1, and when I set 0.2 height and maximum variation to 0.1 it turns 0.3mm or 0.1mm in all the layers,and this is a bad thing.

[image]

### Describe the solution you'd like

I would like to have a minimum adaptive height to do all my print with 0.2 and a few layers with 0.1,And A option to set a specific height or layer to do adaptive.

### Describe alternatives you've considered

--

### Affected users and/or printers

Everyone who want to set a specific part with a lower height ou higher seight of layer in a specific part

### Additional information & file uploads

It will increase mora quality to all my prints,and It would take fastier job

fvrmr, 2021-05-20:
Hi @salvatore54 thank you for your feature request.
This is a duplicate of https://github.com/Ultimaker/Cura/issues/8174. 
That's why I will close this issue. Hope you understand.

salvatore54, 2021-05-20:
That's not exactly the same thing, there's another request.
I would like to have a control of witch layers I want to do the adaptive. Example: "Adaptive layers just between layers 30 and 60" .

fvrmr, 2021-05-20:
Ah that was not clear to me. I will discuss it with the team. Keep you posted!

salvatore54, 2021-05-20:
Ok! 😄 ,Because there's no feature to set for example,adaptive layers just in layer 70 to 200

fvrmr, 2021-05-25:
Hi @salvatore54 I have discussed it with the team. And decided to defer this feature request, it is not a priority for us to add now.
Hope you understand.

Ghostkeeper, 2021-06-02:
> 2 And another problem in adaptive,I just have Maximum variation,but I don't have a minimum control,
because sometimes I want to print, all the layers with 0.2,and just a few parts with 0.1, and when I set 0.2 height and maximum variation to 0.1 it turns 0.3mm or 0.1mm in all the layers,and this is a bad thing.

For this one you do get control, via the Adaptive Layers Topography Size setting.

BloodyRain2k, 2021-06-18:
> For this one you do get control, via the Adaptive Layers Topography Size setting.

Which are pretty cryptic if you expect control in the way of "min layer height, max layer height".