Ultimaker/Cura #9874: Unnecessary movements when printing thin object "askew"

spixe5054, 2021-05-19:
### Application Version

4.9.1

### Platform

Windows 10

### Printer

Ender 3 SKR RPI

### Reproduction steps

[Documents.zip](https://github.com/Ultimaker/Cura/files/6511075/Documents.zip)
Open project
Open settings
Slice objects
Jump to layer 3 and about 75% in the visualization 
(Or print the object and wait for layer 3)

### Actual results

The nozzle jumps back and forth from the printing area to the beginning/end of the object

Video 1 (first half of the object):
https://photos.app.goo.gl/iD7dSBc8tEi44PiQ8
Video 2 (second half of the object):
https://photos.app.goo.gl/FQLaWg2GzoUhV5Y37

### Expected results

The nozzle should stay at the printing area. As far as I can tell, there are not being printed anything at one end of the jump.

See video for better illustration. It's difficult to explain.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

The first half of the length was jumping to the beginning, while the last half jumped to the end and back.
All the continuous travel moves ended up overheating all the stepper motors and resulting in one dead motor.

I don't know if this is an issue with 4.9. I have not tested earlier versions. And I know that the object and settings are not optimal for that thin object, but there is still something going on which should not be going on, as far as I can tell. :-)

fvrmr, 2021-05-20:
Hi @spixe5054 thank you for your report.
There are a few solutions for this issue. 
 1. Slicing this file in Arachne, which has variable line width.  Read here more about our Arachne Engine: https://github.com/Ultimaker/Cura/releases/tag/4.9-beta
[image]

2. Changing combing mode to All instead of Not in Skin
[image]

3. Rotating that model.
[image]

Let me know what works the best for you.

spixe5054, 2021-05-21:
Hello and thank you for the reply.

I wasn't as much looking for a solution as reportingunintentional behavior in the slicer. :-)

It's not really giving me a lot of faith in the slicer, and it scares me leaving the printer alone for any amount of time.
That being said, I've never had issues before. So could be 4.9.1?

fvrmr, 2021-05-28:
It has to do with the slicer algorithm this is changed in Arachne. 
I tried it in 4.8 and it also happened there to with this model.

Ghostkeeper, 2021-06-02:
This is a duplicate of https://github.com/Ultimaker/Cura/issues/4871. A long-standing issue that would be very hard to solve completely, without taking a completely different approach to filling thin gaps.

It is expected to be fixed by Ultimaker/CuraEngine#1210. We did take a completely different approach there.

closed 2021-06-02 by Ghostkeeper