Ultimaker/Cura #9875: Not able to slice print that has size near to max bed without adhesion option

Hajto, 2021-05-19:
### Application Version

4.9.1

### Platform

Mac OS

### Printer

Ender 3

### Reproduction steps

1. Use Ender 3 profile that ships with Cura
2. Load 3mf project  attached to issue 
3. Make sure that adhesion is not selected
4. Wabam it errors for me
5. Check adhession
6. Wabam it works
 

### Actual results

https://www.youtube.com/watch?v=cf85XT1uSVs
If the adhesion is checked it slices, if it is not, then an error is thrown.

### Expected results

If it fits its slices.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[logs2.log](https://github.com/Ultimaker/Cura/files/6511509/logs2.log)
[CE3_Baza.3mf.zip](https://github.com/Ultimaker/Cura/files/6511510/CE3_Baza.3mf.zip)

fvrmr, 2021-05-20:
Hi @Hajto thank you for your report. 
If you go to the custom settings and change there the Build Plate Adhesion Type to none. It would solve your problem.
By disabling it in the recommended settings it will go to skirt, and that will cause this issue.

Let me know if this makes sense to you.

Ghostkeeper, 2021-06-02:
This issue occurs because if the adhesion method is disabled, the Ender 3 profile still prints a skirt, likely in order to prime the nozzle. The skirt needs to take a certain distance from the model in order to not stick to the model even if the material isn't primed yet. With this added distance around the model, it doesn't fit on the build plate any more.

When the adhesion is enabled, it uses a brim. You've overridden the brim size to be smaller than the skirt distance, and as such it takes less space to print a brim than it takes to print the skirt.

So this is expected behaviour.

To print without adhesion, you'd need to reduce the Skirt Distance setting so that it fits on your build plate, similar to how you reduced the Brim Width.

closed 2021-06-02 by Ghostkeeper