Ultimaker/Cura #9880: Under-extrusion at z-seam when infill intersects with layer start

msteele999, 2021-05-20:
### Application Version

4.9.1

### Platform

Windows 10

### Printer

Sovol SV01

### Reproduction steps

1. Load 40x40x40 cube
2. Slice with any infill that puts infill on the layer start z-seam
3. Print - Image 1 shows under-extrusion at the z-seam
4. Slice with no infill (0%)
5. Print - Image 2 shows pristine z-seam

### Actual results

I can reproduce this on every model I slice

### Expected results

normal extrusion on z-seam

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[PETG_cube-40-40-40_no_infil_no_top_no_bottom.gcode.zip](https://github.com/Ultimaker/Cura/files/6516755/PETG_cube-40-40-40_no_infil_no_top_no_bottom.gcode.zip)
[PETG_cube-40-40-40_15pct_infil_no_top_no_bottom.gcode.zip](https://github.com/Ultimaker/Cura/files/6516758/PETG_cube-40-40-40_15pct_infil_no_top_no_bottom.gcode.zip)
[image]
[image]

msteele999, 2021-05-20:
[cura.log.zip](https://github.com/Ultimaker/Cura/files/6517152/cura.log.zip)

smartavionics, 2021-05-20:
Assuming your retraction settings are good, you should fix this by either turn off combing or set the the max combing distance without retract to something like 10.

PS I have 2 x SV01 printers and get very good quality prints using PLA, PETG and TPU. With PETG I use little or no fan to maximise layer adhesion. Hope this helps.

msteele999, 2021-05-20:
Mark - thank you - I'll try turning off combing or set the distance further
- I currently have it set to within infill

fvrmr, 2021-05-21:
Let us know if combing or setting the max combing distance will solve your problem

closed 2021-06-04 by no-response[bot]