Ultimaker/Cura #9881: bug in thr box settings

kak121, 2021-05-20:
### Application Version

4.9.1

### Platform

windws 10

### Printer

CR-10 S PRO 

### Reproduction steps

[image]

### Actual results

[image]

### Expected results

no glich in the boxes settings 

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[image]

look in the settings boxes

kak121, 2021-05-20:
I hope that the problem was not from me 

please tell me 

I appresiate to fed me bake abuot thes issue 

think you

kak121, 2021-05-20:
hi 

ammmm I think I now way thes happened maybe because my computer language is (Arabic).Arebic wrote from right to left

kak121, 2021-05-20:
It is indeed the problem (Computer language Arabic), I changed it to English and it all Looks good. 

Please fix it

fvrmr, 2021-05-21:
Hi @kak121 thank you for your report.
This is indeed happening because of the compute language. 
It is a duplicate of https://github.com/Ultimaker/Cura/issues/9667 that's why I will close this issue. 
We have a ticket on the backlog for this issue.

closed 2021-05-21 by fvrmr