Ultimaker/Cura #9882: Cura doesn't use the Support X/Y Distance set on individual custom support

5axes, 2021-05-21:
### Application Version

Not related to release exist for a long time but still there in 4.9

### Platform

Windows 10

### Printer

not related to a printer 

### Reproduction steps

Define a custom support
Set a individual Support X/Y Distance
Set an other value for this settings 

### Actual results

Cura will only use the global settings to create the support

### Expected results

As it is possible to define an individual value for every support expecting to be able to modify localy the result. 

Note certainly same type of limitation as this report : https://github.com/Ultimaker/Cura/issues/8082 (CURA-7610 for other developers)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

fvrmr, 2021-05-21:
Hi @5axes thank you for your bug report.
I can reproduce this issue. I will discuss this with the team. Keep you posted!

evtrados, 2021-05-27:
Developers: See internal ticket CURA-8260

Ghostkeeper, 2021-06-04:
The X/Y distance currently functions as a per-object setting like "how much distance should support keep from this model". Since support is not keeping distance from other support meshes, overriding the setting in a support mesh is useless.

5axes, 2021-06-04:
Not sure to understand ... Yes the issue is how to define a customised support  X/Y distance  on a support mesh.  As it is possible to define individualy this distance via a per-object settings it should be possible to use this value and not just the global settings ? We don't talk about distance between support themselve but distance between the customized support and the model.
[image]

Ghostkeeper, 2021-06-04:
Basically there are conflicting requirements here then. The Support X/Y Distance says "all support needs to keep this much distance from the model". The model still has a Support X/Y Distance set to 1.0mm (because the extruder's setting is not specialised for the model), so all support keeps 1.0mm distance from the model. Including that custom support.

What would you expect to happen if you set the Support X/Y Distance for the model to 2.0mm and the Support X/Y Distance for the support mesh to 0.16mm?

5axes, 2021-06-04:
On my point of view if we can define a value on a customized support , this value should be the final distance between the Model and this particular support.   If you can specify a value ( here it's the X/Y Distance)  this value should be used to calculate the support according to your local modification and the other support will be calculated according to the general settings.

Ghostkeeper, 2021-06-14:
Similarly, if we can define a value on a single model, this value should be the final distance between that model and all the support.

You have two distances here. Both can be specialised, for either of the models. Cura says that the value specialised in the individual meshes wins. You say that the value specialised in support meshes should win. In reality, they are conflicting constraints.

The conflict is currently resolved in the way that I would expect, but not in the way that you would expect. There is a ticket for it now though, so another developer should eventually look at it to see whom they agree with.

closed 2024-12-09 by github-actions[bot]