Ultimaker/Cura #9883: Solid color being sliced into alternating color layers

majicateball, 2021-05-21:
### Application Version

4.9.0

### Platform

Windows 10

### Printer

Bibo 2 

### Reproduction steps

Created a simple, two-color wheel design in TinkerCad and exported the two colors as separate .stl files. Imported them into Cura. Selected the appropriate material/color/extruder for each part. Merged models. Sent to print. First one came out well for the piece in question (a colored ring on the hub to accent the wheel. The first print, this part came out well, and I noticed an issue with my model for the accent piece around the inner edge of the wheel. I went back to my model in Tinkercad, and corrected the outer wheel. (I did not change anything about the accent ring in the center). Reloaded everything into Cura; following the same steps as before. Sent to print without looking at the sliced model too closely. The wheel came out great, except the center accent piece looked as if the two colors had mixed, but not by printing error. I went back and looked layer by layer in Cura, and the model was actually sliced with the black and green material alternating layers. The model does not have these parts intersecting. I even tried to import the accent ring as its own .stl file, and it still happens. To make it even stranger, another wheel design with a similar center accent piece, sliced perfectly as it should.
I am trying to print this set of wheels for a friend, and I would really appreciate help fixing this.
[image]
[image]
[image]

### Actual results

Some models slice correctly, some have this "layering" effect. It even happens on other parts of the models, but in areas where it doesn't create a visual error, so I've pretty much ignored those. 

### Expected results

[image]

[Project file.zip](https://github.com/Ultimaker/Cura/files/6522019/Project.file.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Please let me know if any more info is needed. I think I included everything that was asked for with this instance. It wasn't a crash, so I didn't include a log file. I included multiple screenshots so you can see the center piece at different layers, some where it is green, and others where it is black. Thank you in advance for your asistance!

fvrmr, 2021-05-21:
Hi @majicateball thank you for your report.
Could you share a project file. (file - save project) 
Your 3mf doesn't have your settings and printer. Slicing the with my own settings I see no problems. 
[image]

Ghostkeeper, 2021-06-04:
In the 3MF model that was given, it's clear to see that the two meshes do overlap, regardless of what Majicateball claims.

Disabling the "Alternate Mesh Removal" setting will remove this effect then.

majicateball, 2021-06-04:
Thank you for your assistance... I hadn't noticed that the one STL files
had that portion of the model on it as well, and thus, didn't realize there
was overlap. Sorry to waste your time on such a simple explanation, but
very grateful for your efforts

closed 2021-06-04 by fvrmr