Ultimaker/Cura #9885: Cura 4.9.1 ignores multiple start code commands

korgano, 2021-05-21:
### Application Version

4.9.1

### Platform

Windows 10

### Printer

Bibo2

### Reproduction steps

1. Load latest BIBO2 definitions into Cura 4.9.1
2. Load model into Cura
3. Slice
4. Save G-Code on SD card
5. Load G-Code into BIBO2 printer
6. Initiate 3D printing

### Actual results

3D printing of the sliced objects begins immediately, with occasional blobs of filament being deposited on the skirt or the object due to the lack of an initial purge line.

### Expected results

Extruders move to the front center of the build plate, deposit three lines of material to clear the nozzle and wipe off any deposits of filament on the outside of the nozzle, then proceed to printing the objects.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Project file and BIBO2 definitions: https://www.mediafire.com/folder/97ja3fhy8799o/Cura_Troubleshooting

Relevant sections of code ignored:
`";Startcode BIBO printers\nM109 T0 S170  ;preheat the other extruder, so it will not knock or ruin the print\nG90 ; absolute mode\nG21 ; metric values\nM82 ; Extruder in absolute mode\nM107\nG28\nG1 Z2 F400\nT0\nG90\nG92 E0\nG28\nG1 Y0 F1200 E0\nG92 E0\nT1\nG92 E0\nG1 X-15.0 Y-92.9 Z0.3 F2400.0\t\t; move to start-line position\nG1 X15.0 Y-92.9 Z0.3 F1000.0 E2\t\t; draw 1st line\nG1 X15.0 Y-92.6 Z0.3 F3000.0\t\t; move to side a little\nG1 X-15.0 Y-92.6 Z0.3 F1000.0 E4\t\t; draw 2nd line\nG1 X-15.0 Y-92.3 Z0.3 F3000.0\t\t; move to side a little\nG1 X15.0 Y-92.3 Z0.3 F1000.0 E6\t\t; draw 3rd line\nG1 X15.0 Y-92 Z0.3 F3000.0\t\t; move to side a little\nG1 X-15.0 Y-92 Z0.3 F1000.0 E8\t\t; draw 4th line\nG1 X-16.0 Y-91.7 Z0.3 F3000.0\t\t; move to side a little\nG1 X16.0 Y-91.7 Z0.3 F1000.0 E10\t\t; draw 5th line\nG1 X16.0 Y-91.4 Z0.3 F3000.0\t\t; move to side a little\nG1 X-16.0 Y-91.4 Z0.3 F1000.0 E12\t\t; draw 5th line\nG1 E11.5 F2400\t\t\t\t; retract filament 0.5mm\"`

Cura also seems to ignore the end code from the definitions in favor of its own end code that is functionally identical.

fvrmr, 2021-05-21:
Hi @korgano thank you for your report.
Is the start gcode in the definition file the same as when you go the the machine settings tab?
Can you also share your gcode?

korgano, 2021-05-21:
Checking the machine settings tab reveals that Cura is not reading the start or end G-Code from the single extruder definitions:

[image]
[image]
[image]

Here's a sample of G-Code that corresponds with the linked project file:
http://www.mediafire.com/file/cupnkqvh6vwbp47/B2DO_BIBO2_Door_Catch.gcode

GregValiant, 2021-05-21:
Where did you get those definition files from?  Were they from Bibo?
You loaded the definitions but did you install a new printer with those definitions?
The start gcode in the Door_Catch.gcode file is right out of the supplied Bibo2_dual.def.json file that comes with Cura.

The Definition file you uploaded (Bibo2 dual.def.json) shows the extruders as:
"machine_extruder_trains":
        {
            "0": "BIBO1",
            "1": "BIBO2"
        },
But the extruder files you uploaded are "Bibo single E1.def.json" and "Bibo single E2.def.json".
I'm not a Python guy but it looks to me like the problem is in your "latest" definition files.

korgano, 2021-05-21:
These are the officially provided definitions.

That said, when using the "install a new printer" option, only the first one (Bibo2 dual.def.json) appears to be visible.

korgano, 2021-05-21:
Repasting the single extruder definitions into the Cura definitions file has made them visible, but every attempt to add them fails. Cura closes or crashes and reopening leaves me with the dual extruder definition.

GregValiant, 2021-05-21:
We've officially left my area of competence, but it would appear that there is something wrong in the extruder definition files as well.

fvrmr, 2021-06-03:
The files BIBO2 single E1.def.json and BIBO2 single E2.def.json are not definition files for extruders but are printer definitions. 
See in the file: `"inherits": "fdmprinter"` instead of `"inherits": "fdmextruder"`

Ghostkeeper, 2021-06-04:
Indeed. Like GregValiant said, the `BIBO1` and `BIBO2` definitions don't exist since the files have different names. After renaming the extruder definition files it still fails because they inherit from `fdmprinter`, not `fdmextruder`, and then extruder-only settings such as the extruder start g-code are undefined. They are defined [in fdmextruder](https://github.com/Ultimaker/Cura/blob/2f8f98fcca7b2fc732538854c6b136720c31d6c3/resources/definitions/fdmextruder.def.json#L78). This is what's causing your issue.

Please repair those definition files, or use the ones built in in Cura. I don't see how the definitions that Bibo provided could ever have worked, so most likely they didn't test it well.

closed 2021-06-04 by Ghostkeeper