Ultimaker/Cura #9886: Crashes at "initalizing build voume" - "Cura has encountered something that doesn't seem right"

Vantage-DS, 2021-05-21:
### Application Version

4.9.0. and 4.9.1

### Platform

Windows 10

### Printer

N/A

### Reproduction steps

Opened Cura using the start menu shortcut, or opened an STL with Cura.

### Actual results

Loading screen stops at "Initializing Build Volume".
Error appears titled "Cura can't start".
Error says "Oops, Ultimaker Cura has encountered something that doesn't seem right.

### Expected results

Cura opens and works.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

**Error Traceback**

Traceback (most recent call last):
  File "__startup__.py", line 74, in run
  File "Console.py", line 36, in run
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\bin\cura_app.py", line 241, in <module>
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\cura\CuraApplication.py", line 829, in run
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\cura\API\__init__.py", line 54, in initialize
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\cura\API\Account.py", line 110, in initialize
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\cura\OAuth2\AuthorizationService.py", line 235, in loadAuthDataFromPreferences
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\cura\OAuth2\AuthorizationService.py", line 78, in getUserProfile
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\cura\OAuth2\AuthorizationService.py", line 117, in _parseJWT
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\cura\OAuth2\AuthorizationService.py", line 257, in _storeAuthData
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\cura\OAuth2\AuthorizationService.py", line 78, in getUserProfile
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\cura\OAuth2\AuthorizationService.py", line 117, in _parseJWT
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\cura\OAuth2\AuthorizationService.py", line 257, in _storeAuthData
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\cura\OAuth2\AuthorizationService.py", line 78, in getUserProfile
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\cura\OAuth2\AuthorizationService.py", line 117, in _parseJWT
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\cura\OAuth2\AuthorizationService.py", line 257, in _storeAuthData
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\cura\OAuth2\AuthorizationService.py", line 78, in getUserProfile
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\cura\OAuth2\AuthorizationService.py", line 117, in _parseJWT
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\cura\OAuth2\AuthorizationService.py", line 257, in _storeAuthData
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\cura\OAuth2\AuthorizationService.py", line 78, in getUserProfile
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\cura\OAuth2\AuthorizationService.py", line 117, in _parseJWT
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\cura\OAuth2\AuthorizationService.py", line 257, in _storeAuthData
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\cura\OAuth2\AuthorizationService.py", line 78, in getUserProfile
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\cura\OAuth2\AuthorizationService.py", line 117, in _parseJWT
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\cura\OAuth2\AuthorizationService.py", line 257, in _storeAuthData
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\cura\OAuth2\AuthorizationService.py", line 78, in getUserProfile
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\cura\OAuth2\AuthorizationService.py", line 117, in _parseJWT
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\cura\OAuth2\AuthorizationService.py", line 257, in _storeAuthData
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\cura\OAuth2\AuthorizationService.py", line 78, in getUserProfile
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\cura\OAuth2\AuthorizationService.py", line 117, in _parseJWT
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\cura\OAuth2\AuthorizationService.py", line 257, in _storeAuthData
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\cura\OAuth2\AuthorizationService.py", line 78, in getUserProfile
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\cura\OAuth2\AuthorizationService.py", line 117, in _parseJWT
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\cura\OAuth2\AuthorizationService.py", line 257, in _storeAuthData
[... 1056 lines cut ...]
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 124 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00004544 (most recent call first):
  File "threading.py", line 302 in wait
  File "threading.py", line 433 in acquire
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 124 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00004cdc (most recent call first):
  File "threading.py", line 302 in wait
  File "threading.py", line 433 in acquire
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 124 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Current thread 0x000054d0 (most recent call first):
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\cura\CrashHandler.py", line 396 in _logInfoWidget
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\cura\CrashHandler.py", line 182 in _createDialog
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\cura\CrashHandler.py", line 86 in __init__
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\bin\cura_app.py", line 167 in exceptHook

fvrmr, 2021-05-21:
Hi @Vantage-DS thank you for your report. 
This looks familiar as https://github.com/Ultimaker/Cura/issues/9861
Could you try to remove the 4.9 folder here: C:\Users\<your username>\AppData\Roaming\cura
That will reset your preferences.

Just to be sure make a backup of your Cura so you won't loose your profiles and settings. 

Let me know if that works for you.

Vantage-DS, 2021-05-21:
That does work and is the fix I've been doing, but it probably happens about once maybe twice a week. I initially thought 4.9.1 fixed it but it did it again today :(

Rygura, 2021-05-23:
```
2021-05-23 18:22:03,048 - WARNING - [MainThread] cura.OAuth2.AuthorizationHelpers.parseJWT [123]: Could not retrieve token data from auth server: {"errors": [{"id": "35bfda71-c84d-4f9c-b985-b14a2e0ed25d", "code": "tokenExpired", "http_status": "403", "title": "The given token has expired.", "meta": {"error_type": "AuthError", "url": "https://account.ultimaker.com/check-token"}}]}
2021-05-23 18:22:03,050 - DEBUG - [MainThread] cura.OAuth2.AuthorizationHelpers.getAccessTokenUsingRefreshToken [61]: Refreshing the access token for [https://account.ultimaker.com]
2021-05-23 18:22:03,690 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService._storeAuthData [250]: Attempting to store the auth data for [https://account.ultimaker.com]
2021-05-23 18:22:03,692 - DEBUG - [MainThread] cura.OAuth2.AuthorizationHelpers.parseJWT [114]: Checking the access token for [https://account.ultimaker.com/check-token]
2021-05-23 18:22:04,335 - WARNING - [MainThread] cura.OAuth2.AuthorizationHelpers.parseJWT [123]: Could not retrieve token data from auth server: {"errors": [{"id": "6ba1794c-8351-4270-9aa7-44f7b6975e6e", "code": "tokenExpired", "http_status": "403", "title": "The given token has expired.", "meta": {"error_type": "AuthError", "url": "https://account.ultimaker.com/check-token"}}]}
2021-05-23 18:22:04,337 - DEBUG - [MainThread] cura.OAuth2.AuthorizationHelpers.getAccessTokenUsingRefreshToken [61]: Refreshing the access token for [https://account.ultimaker.com]
2021-05-23 18:22:05,029 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService._storeAuthData [250]: Attempting to store the auth data for [https://account.ultimaker.com]
2021-05-23 18:22:05,031 - DEBUG - [MainThread] cura.OAuth2.AuthorizationHelpers.parseJWT [114]: Checking the access token for [https://account.ultimaker.com/check-token]
2021-05-23 18:22:05,674 - WARNING - [MainThread] cura.OAuth2.AuthorizationHelpers.parseJWT [123]: Could not retrieve token data from auth server: {"errors": [{"id": "1029cea1-8706-49d3-89f7-567d3f3478d7", "code": "tokenExpired", "http_status": "403", "title": "The given token has expired.", "meta": {"error_type": "AuthError", "url": "https://account.ultimaker.com/check-token"}}]}
2021-05-23 18:22:05,676 - DEBUG - [MainThread] cura.OAuth2.AuthorizationHelpers.getAccessTokenUsingRefreshToken [61]: Refreshing the access token for [https://account.ultimaker.com]
2021-05-23 18:22:06,300 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService._storeAuthData [250]: Attempting to store the auth data for [https://account.ultimaker.com]
2021-05-23 18:22:06,303 - DEBUG - [MainThread] cura.OAuth2.AuthorizationHelpers.parseJWT [114]: Checking the access token for [https://account.ultimaker.com/check-token]
```

I'm having the same issue and see the above in the stderr.log file

Deleting the `ultimaker_auth_data` value in the `cura.cfg` file allows the application to start

fieldOfView, 2021-05-24:
Duplicate of #9847 fixed by #9903

nallath, 2021-05-25:
The fix has been merged, so I'm closing this. Thanks to @fieldOfView!

closed 2021-05-25 by nallath