Ultimaker/Cura #9887: Quality profiles based of percentiles of nozzle diameter

brunoosti, 2021-05-22:
### Is your feature request related to a problem?

With not much experience I am creating a custom printer, simplifying an artillery sidewinder x1 definitions and quality files and was experimenting with setting some quality profiles as a percentile of the nozzle diameter (lets say for a draft option: [values] layer_height = =machine_nozzle_size*0.8). But cura won't change the quality profile based of the nozzle size I input into printer settings, but only from the "base" definition file (where que quality profile points to, of course). If the default_value is set to 0.4mm there, it doesn't matter if I type 1mm in the printer settings window. 

### Describe the solution you'd like

I'm not sure if there would be a solution, but shouldn't layer_height = =machine_nozzle_size*0.8 simply relate to the value I have inside the printer settings window, like "line_width" does? If I change the nozzle size, line width follows, as expected. 

### Describe alternatives you've considered

None. I'm not even sure if I'm posting this at the right place. Sorry... : (

### Affected users and/or printers

Could affect everyone that might want to have a quality setting based of the nozzle diameter. 

### Additional information & file uploads

My custom files for an "artillery_b" printer
[artillery_b.zip](https://github.com/Ultimaker/Cura/files/6525593/artillery_b.zip)

Thank you very much for your help, or pointing me to the right direction! : )

brunoosti, 2021-05-22:
another strange part of this is how cura display the result: 
[image]

rburema, 2021-05-28:
@brunoosti Layer height is one of a few things that is determined _before_ the 'quality' and 'intent' layers are added to the system. A certain layer height can be present in multiple different qualities, but a quality is assumed to have only one possible layer height, unless the user overwrites that explicitly in the settings.

There is (might be) a lot of code in the system that assumes it's easy to derive a 'standard' layer height for any given quality. (Not to mention the interface, take a look at the 'Recommended' (as opposed to 'Custom') interface ... the line with radio-buttons up top doesn't select layer height directly, it selects quality!)

brunoosti, 2021-05-28:
Sure. I understand the limitation on the code and how cura might expect we create quality profiles. After all, a profile is highly dependent on layer height. A thicker extrusion would require adjustments all over the place. But, being lazy (hehe) I wondered if I could create a "dynamic" quality profile base of nozzle diameter, and have settings change depending on the nozzle I'm using.

rburema, 2021-06-03:
I'm sorry, but we don't have that currently. Like you said, 'a thicker extrusion would require adjustments all over the place', so 'layer-height' would have to be in all of those formula's it's currently not in. ... and this might be different for different types of printers _anyway_ so you'd loose a lot of the benefit.

I'll close this now, since I don't see us doing this any time soon.

closed 2021-06-03 by rburema