Ultimaker/Cura #9888: how to add material file

zhpt, 2021-05-22:
### Is your feature request related to a problem?

Hi, I want to add a new materials to Cura. But follow this page  https://github.com/Ultimaker/Cura/wiki/Adding-new-machine-profiles-to-Cura without it.  Where should I upload the new materials file or what should I do?
Thank you.

### Describe the solution you'd like

upload or save the new materials to Cura.

### Describe alternatives you've considered

no

### Affected users and/or printers

this materials is apply to our machine and we will pull request later for machine profiles

fvrmr, 2021-05-25:
Hi @zhpt you can add your materials also by a pull request. [Here](https://github.com/Ultimaker/Cura/tree/master/resources/quality/ultimaker3) you can find some examples of the materials. 
I will close this issue. But let me know if you have more question.

closed 2021-05-25 by fvrmr

zhpt, 2021-06-07:
[image]

Hi, I want to add TPR_Material.xml.fdm_material file to Cura, such as the picture, where should I 
upload the file called TPR_Material.xml.fdm_material.

Thank you for your reply .

konskarm, 2021-06-07:
@zhpt Materials are handled in the repository https://github.com/Ultimaker/fdm_materials/ so make sure to open up a pull request there for your .xml.fdm_material profiles.