Ultimaker/Cura #9889: Strong vase mode with z travel up & down along the layers

wasabiguy, 2021-05-22:
### Is your feature request related to a problem?

I love vase mode but the layer bond  & strength is always a problem..

### Describe the solution you'd like

Hi...1st time post here... I love vase mode but the layer bond is always a problem.. curious if it's possible to print with z-axis moving up & down along the layer in cura & hopefully strengthen it..
*Photo example of a cylinder isometric view & side view

Thanks in advance

[image]
[image]

### Describe alternatives you've considered

Can be wavy or zig zags along the layers as long a the bond is not straight exposing the weak point..

### Affected users and/or printers

Anyone who uses vase mode

GregValiant, 2021-05-22:
There would still be layers even if it could be sliced this way.  The shape of the nozzle would come into play leaving radii in the cross section of the extrusion.  You would need some sort of adaptive layers to build up to the deltaZ (a print couldn't start out this way).
It's easier to push the line width to 1.5 x nozzle diameter.
Silky PLA would still have a real problem with layer adhesion - but it looks so darn good.

fvrmr, 2021-05-25:
Hi @wasabiguy thank you for your feature request.
It would be very difficult to implement because of how our slicing algorithm works. So for now I will defer this issue, since it is not a priority for us to add this right now.
Hope you understand!