Ultimaker/Cura #9890: Experimental Bridging not in Walls on Bridge Layer 2 and 3

AcJoker2, 2021-05-22:
### Application Version

4.9.0

### Platform

Windows 10

### Printer

Ender 3

### Reproduction steps

I've configured the experimental bridge setting as shown here.
I sliced several "bridges" with it.
Before that I tested all bridge settings from min to max, but didn't see any influence on the problem.
Either it is a bug or there are other settings that require this, but which are not documented in the context. Or I couldn't find the info.
[image]

### Actual results

The walls (inside and outside) of bridge layer 2 and 3 are not printed at reduced speed, but at normal wall speed.
In the code, the walls are also declared as normal walls and not as bridges.

In addition, the speed of the walls in the 1st bridge layer is displayed as a speed gradient.
According to the code, the speed is constant at the level of the bridge speed.
So it just seems to be a display bug.

But this could be related to the first error and trigger an incorrect bridge detection for layers 2 and 3.

I also don't understand why more material is extruded in the 1st layer for a short time at the end of the walls.

[image]
[image]
[image]

Layer 50 and 51 (Bridge Layer 1 and 2)

[Code.txt](https://github.com/Ultimaker/Cura/files/6526313/Code.txt)

### Expected results

I would expect that the walls will also have a reduced bridge speed, and that in all 3 bridge layers.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Project File
[Bridge 100mm.zip](https://github.com/Ultimaker/Cura/files/6526331/Bridge.100mm.zip)
Log
[cura.log](https://github.com/Ultimaker/Cura/files/6526339/cura.log)

smartavionics, 2021-05-22:
The walls on bridge layers 2 and 3 use the normal wall settings. That is by design.

The change in speed/flow at the end of a bridge wall is to compensate for the underextrusion that occurs following the bridge wall. My Cura builds (see https://github.com/smartavionics/Cura/releases) feature an extra setting (Bridge Wall End Boost) that controls that feature but, unfortunately, it never became part of UM Cura.

AcJoker2, 2021-05-22:
Thanks for the explanation. I just don't understand why the walls in layers 2 and 3 are not also declared as bridges.

Should the firmware not regulate the extrusion, at least with LA / PA.
From my point of view, it is very disadvantageous if the slicer intervenes in this way and this cannot be switched off.

So it's not a bug after all, sorry.

closed 2021-05-22 by AcJoker2