Ultimaker/Cura #9891: Hint overides button for pre-heat extruder

tamarit77, 2021-05-22:
### Application Version

4.9.1

### Platform

Windows 10

### Printer

Hellbot Magna Dual

### Reproduction steps

1. Position mouse cursor over Extruder Preheat button on Supervising Tab.
2. Hint appear and blink overriding the button. 

### Actual results

Hint is set over the button and button can not be pressed.  I need to press mouse button quick and repetedly until suddenly action ocurrs.

### Expected results

Hint can appear in other position than over the button, or dissapear once clicked, or don't appear all the times I want to preheat.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

https://user-images.githubusercontent.com/75040662/119236829-df49ee80-bb0f-11eb-9167-9ce0748b3e5c.mp4

fvrmr, 2021-05-25:
Hi @tamarit77 thank you for your report.
This is happening on the monitor tab right? And does it happen on other buttons as well?

tamarit77, 2021-05-25:
Hi,
Only in the monitor tab, and only with Extruder 1 and Extruder 2 pre-heat.
Not happening on Build Plate pre-heat.

 cura hint on top button.mp4
<https://drive.google.com/file/d/1_O6D74M4e_tjOdLcKbm2ltykJV6K_l86/view?usp=drive_web>

fvrmr, 2021-05-28:
Can you also share your log file? You can find your log file here: `C:\Users\\<your username>\AppData\Roaming\cura\<Cura version>\cura.log`

tamarit77, 2021-05-28:
Sure, there it goes.

tamarit77, 2021-05-28:
[cura.log](https://github.com/Ultimaker/Cura/files/6561504/cura.log)

Done

nallath, 2021-06-01:
This is fixed for the next release. Thanks for reporting it!

commit 4f49a381 (2021-06-01): Fix pre-heat button being ocluded by tooltip

Fixes #9891
closed 2021-06-01 by nallath