Ultimaker/Cura #9892: Import Material

BOFHELL, 2021-05-23:
### Application Version

4.9.1

### Platform

Mac OSX High-Sierra

### Printer

doesn't matter

### Reproduction steps

1. Start Cura
2. Open material settings
3. import 1st profile
4. try to import 2nd profile  -> open dialog not appears

### Actual results

no import dialog

### Expected results

open import dialog, again

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

installed plugins:
octoprint, mesh tool

fvrmr, 2021-05-25:
Hi @BOFHELL thank you for your report. 
Can you share your profiles you want to import? And your log file? You can find your log file here: 
`$USER/Library/Application Support/cura/<Cura version>/cura.log`

BOFHELL, 2021-05-26:
sure. cura log and one profile attached.

i have to restart cura to import additional profiles. all profiles can be imported.
after importing the "import" button has no function...

[cura.log](https://github.com/Ultimaker/Cura/files/6548808/cura.log)

[Extrudr_PLA_BlueFire.xml.fdm_material.zip](https://github.com/Ultimaker/Cura/files/6548814/Extrudr_PLA_BlueFire.xml.fdm_material.zip)

fvrmr, 2021-05-31:
Hi @BOFHELL thank you. It works for me. 
It happens for every printer selected? And when you first import your materials do they work?

BOFHELL, 2021-06-02:
have just 1 printer.
i can import all my profiles and materials. but for all its the same issue. it just worked once and after import success i've to restart cura.

OS: macOS High Sierra 10.13.6

fvrmr, 2021-06-03:
I can't see anything why it is behaving like this in the logs. And I can't reproduce this either.
So for now we have to wait if other users have the same issue to have more info than we now have. 

Luckily you can import at least, but I understand that it is annoying that you can't import after importing.

BOFHELL, 2021-06-06:
i fixed it by copy cura app to /Applications
now all worked fine.   thx 4 help

closed 2021-06-06 by BOFHELL

fvrmr, 2021-06-07:
Thanks for letting me know!