Ultimaker/Cura #9893: Ryzen 9 5950X Cura shows white screen and does not start up, when AMDs SMT is activated in UEFI

L3mming, 2021-05-23:
### Application Version

4.9.1

### Platform

Windows 10

### Printer

Creality CR-10S Pro

### Reproduction steps

1. I changed the CPU from an Intel Core i7 4970K with an ASUS Z97 Extreme 6 to an AMD Ryzen 9 5950X with an ASUS ROG Strix X570-F Gaming. Before changing, all old drivers were uninstalled
2. Cura 4.9.0 did not start, only showed a white screen and Windows 10 reports, that the program is not responding, so I updated to 4.9.1, no effect.
3. As other software also didn't work properly, I thought that maybe SMT would be a good starting point to look for issues, so I deactivated it. That solved the problem.
4. To be sure, I reactivated it again and voilà, the problem reappeared.
5. Deactivated SMT once again --> Cura worked again
6. Reactivated SMT --> Cura not working.

### Actual results

After deactivating SMT in the UEFI, Cura works as expected, no white screen anymore.

### Expected results

Nothing should have happened by deactivating SMT, however, it did solve the problem.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

No project file necessary, as it is a general problem.
Log file is attached.
[cura.log](https://github.com/Ultimaker/Cura/files/6528006/cura.log)

fieldOfView, 2021-05-25:
> Cura won't run on computers with a Ryzen graphic card.

That is painful to read. Especially since Ryzen is not a "graphic card", but a microprocessor.

L3mming, 2021-05-25:
Hi fvrmr, I don't use a Ryzen graphics card, the Ryzen 9 5950X doesn't even have a graphics card, as it is not an APU, so this can't be the issue.

fvrmr, 2021-05-25:
Thank you for explaining it to me. Still learning every day.  Then indeed it is not the case for you. 
I will put this up for discussion with the team. Keep you posted!

fvrmr, 2021-06-01:
I have discussed it with the team, and this is not something to fix from our side. 
You could contact AMD Ryzen about this, since this is also affecting other software. 
I will close this issue, hope you understand.

closed 2021-06-01 by fvrmr

L3mming, 2021-06-01:
Thank you for discussing it with your team. However, deactivating SMT only fixed the issues for Cura, the other software (Z88Arion V3) was unaffected but that so I still think Cura and SMT don't like each other right now.

L3mming, 2021-06-01:
Sorry for the hassle, the problem was that the Intel drivers were not correctly uninstalled by the system before I changed the mainboard and the CPU (it seems that the uninstallers do not uninstall all the chipset drivers). Old CPU was still in the device manager list, so I manually deleted all the drivers using the device manager, now Cura is working again. Why deactivating SMT fixed this issue is weird though. Topic closed from my side as well.

fvrmr, 2021-06-03:
Hi @L3mming good to know that Cura is now working again!