Ultimaker/Cura #9895: Add per printer flow rate (mm)3/s limits, warnings, and calculations to print

tastyratz, 2021-05-23:
### Is your feature request related to a problem?

Printer extruders/hot ends/etc. have flow rate limits. It's easy to exceed these with different speeds, layer heights, or line widths.

### Describe the solution you'd like

Printer definitions should allow for a maximum flow rate per printer in the profile.

Per print configuration should calculate expected typical/peak flow rates even if done in a simple way like shown here:
https://polygno.com/flow_rate_calculator

If any of those 3 settings are configured/calculated to exceed the set limits, a warning pop/yellow box around the setting should pop (i.e. "warning: configured layer height would result in flow rate of 30 (mm)3/s which exceeds selected printer limit of 17 (mm)3/s)

### Describe alternatives you've considered

Manually doing the math the long way or waiting till I grind my filament to dust making a failed print

### Affected users and/or printers

everyone

fvrmr, 2021-05-25:
Hi @tastyratz thank you for your request.
In our master we've added flow visualisation, would that work also for you? Let me know what you think.

tastyratz, 2021-05-26:
@ fvrmr I'm not sure, I am not able to find any documentation for it or find anything as far as views go within Cura? Is this something I can access today?

If it's just part of a layer view, I feel like that doesn't really pull it forefront for a warning. It seems like a simple calculation like other calculated values to include as a flag?

My thoughts are it would still add value to prevent a possible failure.

tastyratz, 2021-05-28:
https://help.prusa3d.com/en/article/max-volumetric-speed_127176/

Apparently, Prusaslicer has a maximum volumetric speed function which I did not realize till last night, I found it when I was trying to manually calculate flow rates to make a test tower and benchmark my filament. It was mentioned in the flow calculator thread https://www.reddit.com/r/3Dprinting/comments/jzkxpa/handy_flow_rate_calculator/ so I took a look.
Something similar sounds phenomenal! Reviewing the documentation makes me believe it might actually make sense to set in material profiles instead or in addition.

fvrmr, 2021-06-01:
I have discussed it with our team and this is a duplicate of https://github.com/Ultimaker/Cura/issues/5248
I will close this issue, but I will create a ticket on our backlog for this feature request and will link the other issue in that ticket, so we keep everything in one place.

closed 2021-06-01 by fvrmr