Ultimaker/Cura #9897: 4.9.1 freezes while "Initializing build volume"

addictionInk, 2021-05-23:
### Application Version

4.9.1

### Platform

Windows 10

### Printer

Artillery Sidewinder X1

### Reproduction steps

The last time I used Cura, I closed the program and then shut down my computer. I was gone for a week and when I returned I booted my computer, and attempted to load Cura. It freezes at initializing build volume and will not open the program. I have uninstalled, restarted, and reinstalled but that did not work. 

### Actual results

It still freezes and does not finish loading or open cura.

### Expected results

IT should load and open Cura

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/6528804/cura.log)

nallath, 2021-05-25:
This has been fixed by https://github.com/Ultimaker/Cura/pull/9903 (and is a duplicate of https://github.com/Ultimaker/Cura/issues/9847)

closed 2021-05-25 by nallath

Here-2Learn, 2021-06-28:
This is happening to me as well. Please inform me of the fix

addictionInk, 2021-06-28:
The problem was with my account sign in, to solve this I followed these
steps:

1. Close Cura
2. Search for Cura.cfg in windows explorer. It is usually on your C: drive
under you your_username\%APPDATA%\cura\4.9.
3. Open cura.cfg in notepad
4. Delete the entire row that starts with "ultimaker_auth_data = "
5. Now reopen Cura and sign in

The easiest way I have found is just open windows explorer and use the
search bar type or paste cura.cfg, I then add a shortcut to my desktop
because it happens once a month or more. Hope this helps.