Ultimaker/Cura #9898: Not working on Mac Mojave

thebioguy, 2021-05-23:
### Application Version

4.9.1

### Platform

Mac OS Mojave

### Printer

Neptune 2

### Reproduction steps

Upon launch

### Actual results

“Elegoo Cura” is damaged and can’t be opened. You should move it to the Trash.

### Expected results

Launching the program

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Help

fvrmr, 2021-05-25:
Hi @thebioguy thank you for your report.
Can you share your log file? You can find your log file here: `$USER/Library/Application Support/cura/<Cura version>/cura.log`

tkredbaron, 2021-05-29:
hey,
same problem

thebioguy, 2021-05-29:
The version provided on the included SD card works on Mojave.

fvrmr, 2021-05-31:
I still need more information or a log file. Is the installer working? Or does this happen when you want to open Cura?
Probably the version provided on the SD card is not our version.

Ghostkeeper, 2021-06-04:
Is this also happening for Ultimaker Cura, or only for Elegoo Cura? It could be that Elegoo did something weird when packaging their MacOS release. Packaging for MacOS is pretty difficult after all.

thebioguy, 2021-06-04:
Ultimaker Cura works fine on Mac and is probably not much different than
the Elegoo version. For some reason the version on Elegoo website will not
boot on Mojave, but the version bundles with the printer works fine.

Ghostkeeper, 2021-06-04:
Then we'll have to close this. We didn't create Elegoo's version of Cura, and we can't fix their version for you if they broke something.

In fact, I can't even find their source code. They don't have it in [their Github account](https://github.com/elegoogroup) nor is it downloadable from their website. They might be in violation of Cura's license agreement there...

closed 2021-06-04 by Ghostkeeper

sodawave, 2022-07-31:
you can run this command to fix the error

`xattr -cr /Applications/Elegoo\ Cura.app `