Ultimaker/Cura #9900: Support of "anycubic mega x 3d"

Syu4328, 2021-05-24:
### Is your feature request related to a problem?

Now, we have to edit "i3 mega" configuration to use "mega x". I want you to add an item of "anycubic mega x 3d" in the printer list.

### Describe the solution you'd like

If you add the configuration of "mega x 3d" in the printer list, this software, Cura, should be easier to use than before. 

### Describe alternatives you've considered

Please make the item of "mega x 3d" in the printer list. 

### Affected users and/or printers

Most of the users who use mega x 3d, or the users who will get "mega x 3d" printer will find this software easy to use and will use it a lot. 

### Additional information & file uploads

Not any.

GregValiant, 2021-05-25:
When you add a printer, on the right side of the dialog is a box that allows you to change the name.  Just above that box is a tag that says "author" and in the case of your Mega printer the author hired by AnyCubic is "TheTobby".  That's who wrote the definition file that AnyCubic submitted to Ultimaker for inclusion in Cura.  So you would need to get hold of AnyCubic and ask them if they have the new definition file ready for download.

closed 2021-05-25 by fvrmr

NilsRo, 2021-08-08:
Added Mega X profile to 4.10. Please try this version...

DocMinus, 2021-09-15:
> Added Mega X profile to 4.10. Please try this version...

@NilsRo where? I don't see it, unless you mean i3 Mega X. Is that the same as Mega X ? (noob here, sorry. Tnx in advance).

NilsRo, 2021-09-15:
> > Added Mega X profile to 4.10. Please try this version...
> 
> 
> 
> @NilsRo where? I don't see it, unless you mean i3 Mega X. Is that the same as Mega X ? (noob here, sorry. Tnx in advance).

Yes