Ultimaker/Cura #9901: Custom profiles missing

hyudryu, 2021-05-24:
### Application Version

4.9.1

### Platform

Windows 10

### Printer

CR-10

### Reproduction steps

This issue started happening after upgrading from 4.6.x -> 4.9.0 The issue remained after upgrading from 4.9.0 -> 4.9.1

1. The issue occurs upon opening the app.

### Actual results

When I run Cura, it is missing a lot of my custom profiles. One of my custom profiles was actually loaded by default, but is missing from the drop-down menu. Here's an example of what's missing. I had a profile named "CR10 .6 CF PC" which was selected by default after opening the software, but is nowhere to be found in the drop-down menu, nor in the settings.
[image]
[image]

This is one of many custom profiles that are missing

### Expected results

The expected results would be to have all my custom profiles visible.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/6531827/cura.log)

hyudryu, 2021-05-24:
After closing and re-opening the program, sometimes the custom profiles will magically show up again. This was after closing the program 3 times and opening for the 4th time, Heisenbug perhaps?
[image]

fvrmr, 2021-05-25:
Hi @hyudryu thank you for your report. 
Did it just solve doing a plain restart? Did you maybe change your material? Because profiles are related to materials.

hyudryu, 2021-05-30:
Hi @fvrmr , thank you for getting back to me regarding this issue. I've been using Cura since day 1, and I have not once touched the "materials" selection. I've always stuck with the default material, and just made new profiles for the different filaments I have.

Restarting it is a hit or miss, sometimes it will take multiple restarts for the profiles to appear, sometimes it takes 1 restart. Aside from the restart, I don't do anything else. I usually keep Cura running in the background 24/7, I never really close it out since I slice and print multiple things every week. Sometimes the issue just randomly pops up when I switch over to Cura after a while. 

Please let me know if there's any more information that I can help provide!

fvrmr, 2021-05-31:
Hi @hyudryu thank you.

I think it is a duplicate of https://github.com/Ultimaker/Cura/issues/8229

There you can read that there was a bug in 4.6 with wrong profiles and since you come from 4.6 to 4.9 I think that is why this is happening to you. 

Here is how profiles should work: https://github.com/Ultimaker/Cura/issues/8229#issuecomment-680042094
And here is a work around to migrate your profiles the correct way: https://github.com/Ultimaker/Cura/issues/8229#issuecomment-684799909

Let me know if that works for you!

closed 2021-06-14 by no-response[bot]