Ultimaker/Cura #9902: FR: When scaling, use bottom of object as origin

logiclrd, 2021-05-24:
### Is your feature request related to a problem?

I recently scaled up an object and then printed it. For unrelated reasons, I have the option to force models to snap to the build plate turned off. The scaling was done using the centre of the object as an origin, so that the top went up, and the bottom went down. This resulted in the bottom going beneath the build plate. Sometimes I want this to happen, which is why I explicitly have it configured to not automatically bump the object up, but I don't think I ever want it to happen automatically when scaling. In this case, I didn't notice it had happened, and just printed an object with the bottom ~5mm cut off.

### Describe the solution you'd like

I can think of two possible paths:

* Make scaling, specifically, use an origin with the object's min(Z), rather than its centre Z. Then, whenever scaling an object, its interface with the base plate would remain unchanged. This could be presented as a configurable option: "Scaling Z origin: top/middle/bottom"

* Alternately, an option could be added that, when enabled, ensures that objects never go beneath the build platform as a result of scaling. Scaling works the same way it does now, but then also if "Automatically drop models to build plate" is turned off and the scaling causes the object to extend into negative Z, it is automatically translated up.

### Describe alternatives you've considered

Without support in the software, I am forced to remember to adjust the Z whenever I scale an object.

### Affected users and/or printers

People who scale models and who have Cura configured to allow models to go beneath Z=0.

fvrmr, 2021-05-25:
Hi @logiclrd thank you for your feature request.
I will discuss this with the team. Keep you posted!

logiclrd, 2021-05-28:
Awesome, thanks :-)