Ultimaker/Cura #9905: Backup restore (using AppImage) doesn't restore anything

wernight, 2021-05-25:
### Application Version

4.9.1

### Platform

Ubuntu (Crostini)

### Printer

Sidewinder X1

### Reproduction steps

1. Created a backup from another Ubuntu (Crostini) also from AppImage
2. Downloaded AppImage on a new Ubuntu (Crostini)
3. Launched Cura and passed the login etc.
4. Extension > Backups > Restore (selecting latest 4.9.1 version)
5. Yes to restart

### Actual results

Closes Cura, nothing in the console logs, and when starting Cura again no visible setting were imported.

### Expected results

Restarts Cura and I've again Dark theme, my currency and filament settings with favorites...

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[stderr.log](https://github.com/Ultimaker/Cura/files/6538423/stderr.log)
[stdout.log](https://github.com/Ultimaker/Cura/files/6538424/stdout.log)

Noticed in the stderr:

```
DEBUG - [MainThread] cura.Backups.Backup.restore [158]: Moving preferences file from /home/user/.local/share/cura/4.9/cura.cfg to /home/user/.local/share/cura/4.9/cura.cfg
DEBUG - [MainThread] UM.Qt.QtApplication.windowClosed [475]: Shutting down cura
...
WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [411]: file:///tmp/.mount_Cura.AxoZ7Qn/usr/bin/qml/UM/Preferences/PreferencesDialog.qml:113: TypeError: Cannot read property 'resetEnabled' of null
```

wernight, 2021-05-25:
Looks like the directory is wrong, running this command fixes the restored backup:

```
rm -rf .config/cura/4.9
cp -r .local/share/cura/4.9 .config/cura/
```

wernight, 2021-05-25:
The issue is probably in the move that moves from/to the same location:

https://github.com/Ultimaker/Cura/blob/7e98fb7fb065b6d672ecc808f7747d962c05f681/cura/Backups/Backup.py#L161

I didn't find yet where the path is stored. I noticed that the [appdirs](https://pypi.org/project/appdirs/) package is requested but I didn't find its uses. Given I'm using the official AppImage

fvrmr, 2021-05-25:
Hi @wernight thank you for your bug report.
I can reproduce this as well on Windows. I will discuss this with the team. 
Keep you posted!

fvrmr, 2021-05-28:
I have created a ticket on our backlog for this issue.
Devs see CURA-8276

MariMakes, 2023-05-17:
Quick update from our side 👋

We are cleaning up our open issues, so we can focus on the most requested and needed features and bug fixes from our community and from UltiMaker, I stumbled upon your issue about the backup not working anymore. 

Since this issue is so old we quickly retested if this was still an issue and it seems to have been resolved. 

To me, it doesn't seem like a bug in Ultimaker Cura, since we have a solution in place. I'll be closing this issue. If you still encounter this, please feel free to let us know and we'll take another look. 👍 

Apologies for any inconvenience caused by our delayed response.