Ultimaker/Cura #9906: Attempting to use calculations in Start/End GCode breaks it without warning about it

BloodyRain2k, 2021-05-25:
### Application Version

4.9.1

### Platform

Win 7 x64

### Printer

CR6

### Reproduction steps

Enter characters use in equations into the { } of a placeholder in the Start/End GCode
Close the window and let it re-slice the part
Save the GCode to disk and verify that it failed to replace the placeholders without any indication

[image]

### Actual results

I want to point out that this is even the 2nd attempt where I commented out the line, and it still broke.

[image]

### Expected results

If it fails to fill in the placeholders because there's something in the GCode that breaks it, I expect it to give me a warning at the very least.

But I can do the whole process and nothing out of the ordinary happens.

On a side note: I was hoping it would be possible to use math with the placeholders, but it was only a test for a tiny QoL thing.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[Bearing_Clip_686.zip](https://github.com/Ultimaker/Cura/files/6541951/Bearing_Clip_686.zip)

fvrmr, 2021-05-27:
Hi @BloodyRain2k thank you for your report. 
It is expected behaviour you can't uses formulas in the start g-code, so that's why it isn't working.

fvrmr, 2021-05-31:
Hi @BloodyRain2k I have discussed it with the team and we decided to defer this issue. 
It is not a priority right now to add a warning when you want to use formulas in the start/end gcode.
Hope you understand.

BloodyRain2k, 2021-05-31:
I can understand not wanting to make calculations at this point a priority, but not adding a warning when something broke the parser? That seems bad.

Like when I tried to change my experiment to
```M109 S190
M300 P100 S660
M109 S{material_print_layer_0_temperature}```
I made a typo in the first line and it read `M109 S{190}` instead, which also broke every other placeholder, just like the commented line with the placeholder calculation.

At the very least it should warn if something went wrong while filling in the placeholders.

Ghostkeeper, 2021-06-04:
People add stuff in brackets to their start g-code all the time, especially for firmware like Repetier's. Can't warn them all the time for things that are perfectly valid. So what distinguishes a perfectly valid g-code intended for third party software from an incorrectly formatted setting key intended for Cura to read?

FroggieFrog, 2023-04-21:
This issue is still present with 5.3.1.
A solution could be to ignore commented lines that fail to parse, but show a warning if the line failed to parse but is not commented.

```gcode
; do not show warning AND continue to replace other variables
; no arithmetic in cura
;M190 S{material_bed_temperature_layer_0 * 0.7}

; is still parsed and replaced despite the above statemant
M190 S{material_bed_temperature_layer_0}

; show warning | maybe stop slicing
M190 S{material_bed_temperature_layer_0 * 0.7}
```

closed 2024-11-19 by GregValiant