Ultimaker/Cura #9908: Cura chrashes on Big Sur - 4.9.1 and also 4.9.0

NoTrace-Kai, 2021-05-26:
### Application Version

4.9.0 and 4.9.1

### Platform

Mac OS Big Sur

### Printer

Ender 3 V2

### Reproduction steps

Just when i want to open the Cura App

### Actual results

The Crash-Log is shown

### Expected results

Also after installing Cura 4.9.0 the error is there ...

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

Process:               cura [16276]
Path:                  /Applications/Ultimaker Cura.app/Contents/MacOS/cura
Identifier:            cura
Version:               4.9 (4.9.1)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           cura [16276]
User ID:               501

Date/Time:             2021-05-26 14:33:44.591 +0200
OS Version:            macOS 11.4 (20F71)
Report Version:        12
Bridge OS Version:     5.4 (18P4663)
Anonymous UUID:        D1E738EE-9EE4-35B6-AFEE-2A9E044D9E9E

Sleep/Wake UUID:       1BD190DD-BFBD-4B18-910E-C4A62935B747

Time Awake Since Boot: 16000 seconds
Time Since Wake:       5700 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x00000104bff0d000
Exception Note:        EXC_CORPSE_NOTIFY

VM Regions Near 0x104bff0d000:
    __LINKEDIT                  17428a000-17429a000    [   64K] r--/rwx SM=COW  /Applications/Ultimaker Cura.app/Contents/MacOS/libQt5WebSockets.5.dylib
--> 
    MALLOC_NANO              600000000000-600008000000 [128.0M] rw-/rwx SM=PRV  

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff2056f92e __pthread_kill + 10
1   libsystem_pthread.dylib       	0x00007fff2059e5bd pthread_kill + 263
2   libsystem_c.dylib             	0x00007fff20482be5 raise + 26
3   libsystem_platform.dylib      	0x00007fff205e3d7d _sigtramp + 29
4   ???                           	000000000000000000 0 + 0
[... 1189 lines cut ...]
__UNICODE                          588K        1 
mapped file                      930.9M      539 
shared memory                      764K       17 
===========                     =======  ======= 
TOTAL                              3.2G     2891 
TOTAL, minus reserved VM space     3.1G     2891 

Model: MacBookAir9,1, BootROM 1554.120.19.0.0 (iBridge: 18.16.14663.0.0,0), 4 processors, Quad-Core Intel Core i7, 1,2 GHz, 16 GB, SMC 
Graphics: kHW_IntelIrisPlusGraphicsItem, Intel Iris Plus Graphics, spdisplays_builtin
Memory Module: BANK 0/ChannelA-DIMM0, 8 GB, LPDDR4X, 3733 MHz, SK Hynix, H9HCNNNFBMALPR-NEE
Memory Module: BANK 2/ChannelB-DIMM0, 8 GB, LPDDR4X, 3733 MHz, SK Hynix, H9HCNNNFBMALPR-NEE
AirPort: spairport_wireless_card_type_airport_extreme, wl0: Feb 17 2021 17:25:29 version 16.50.16.0.3.6.100 FWID 01-a25c2f56
Bluetooth: Version 8.0.5d7, 3 services, 27 devices, 1 incoming serial ports
Network Service: Thunderbolt Bridge, Ethernet, bridge0
Network Service: Wi-Fi, AirPort, en0
USB Device: USB 3.1 Bus
USB Device: USB 3.1 Bus
USB Device: Apple T2 Bus
USB Device: Touch Bar Backlight
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Headset
USB Device: Ambient Light Sensor
USB Device: FaceTime HD Camera (Built-in)
USB Device: Apple T2 Controller
Thunderbolt Bus: MacBook Air, Apple Inc., 85.0

jatkinson10, 2021-05-26:
I'm also experiencing the same issues. 4.8 worked flawlessly, however, 4.9.x has been problematic since I updated.

mbourque, 2021-05-26:
I also experience this quite often

fvrmr, 2021-05-28:
Hi @NoTrace-Kai thank you for your bug report.
I will discuss this with the team. Keep you posted!

fvrmr, 2021-05-31:
@NoTrace-Kai I have discussed it with the team and if you have a M1 then we can't really help you right now since we don't support M1. There are incompatibilities between the OpenGL libraries used and the latest firmware used by macbooks. 

If you don't have a M1 we need to investigate this further.

NoTrace-Kai, 2021-05-31:
Hi. 

Thanks a lot for your answer. 
My MacBook Air is a 2020 with Intel i7 CPU, 16GB RAM and 512GB SSD. 

Greetings Kai

fvrmr, 2021-06-04:
Thank you, do you maybe have a second monitor?

NoTrace-Kai, 2021-06-04:
No. Only the one from the MacBook Air.

Ghostkeeper, 2021-06-04:
By the looks of that stack trace, Qt crashed while it was executing some of the Javascript code in the QML files. Qt should never crash, so it is a bug in their side. In the update to Cura 4.9 we updated the version of Qt to a newer version, so this might be a bug that arose in one of the newer versions.

Another possibility is simply that your computer ran out of memory, since the instruction it apparently crashes on is a Malloc (allocating memory). Cura typically uses like 200-500MB memory though, so unless you had some real heavy stuff running in the background that shouldn't have been a problem for your computer.

Assuming the first possibility there, it's possible that there is a workaround for us. Maybe we should avoid using a certain function in Javascript, or a certain type of syntax. However it's pretty much impossible for us to know what Javascript is causing this issue. We haven't been able to reproduce this issue in any of our testing computers. If we have a reproduction, we could try figuring this out with a debugger looking into the memory of Qt to determine which QML file is causing the issue, and perhaps where in that file it's going wrong. But until then, there is little we can hope to achieve here.

Z122-JPL, 2021-11-14:
@NoTrace-Kai , I wonder if this problem could be solved by this solution : 
https://github.com/Ultimaker/Cura/issues/10789#issuecomment-968054449

mbourque, 2024-11-19:
> Is this still an issue in current Cura versions (5.8.0 and up)?  Can this be closed?

Fixed Regards,Michael Bourque

closed 2024-11-19 by GregValiant