Ultimaker/Cura #9909: Cura 4.9.X is crashed when you click on scale

detlefbrand, 2021-05-26:
### Application Version

4.9.0 and 4.9.1

### Platform

Kubuntu 20.4

### Printer

Ultimaker 2+

### Reproduction steps

1. Start Cura
2. load stl file
3. click on object
4. the menu on the left appears
5. click on scaling

### Actual results

Cura crashes

### Expected results

the size of the object can be changed

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

The bug can be reproduced. I installed Kubuntu 20.4 and Cura 4.9.1 on another PC and tested it with the same result (bug)

fieldOfView, 2021-05-26:
Duplicate of #9709

fvrmr, 2021-05-27:
Closing because it is a duplicate.

closed 2021-05-27 by fvrmr