Ultimaker/Cura #9911: Tooltips on hover print settings

skorpi0n, 2021-05-27:
### Is your feature request related to a problem?

Quite a few times, I've forgotten to verify the correct adhesion type, since I thought I had selected the correct one earlier.

### Describe the solution you'd like

I would like to be able to just hover the print settings icons, like Infill and Adhesion and see a tooltip which describes a bit more of the settings used. Like [infill density, infill pattern...], [adhesion type, brim width, brim/skirt line count...]
[image]
I think this would help verifying commonly changed/used settings easier than to go deeper in the settings.

### Describe alternatives you've considered

An alternative would be to have different icons for each adhesion type.

### Affected users and/or printers

I think this will benefit users that make many short testprints, for example while prototyping, and might overlook some settings, because they thought they were correct.

fvrmr, 2021-05-27:
Hi @skorpi0n thank you for your feature request. 
If you have the settings guide plugin installed and hover over infill pattern as example, this is what you'll see: 
[image]

You can download the settings guide plugin in the marketplace. Let me know if that will work for you!

Ghostkeeper, 2021-06-07:
This request is about adding tooltips to the icons in the header bar, not about other tooltips throughout the application.

fvrmr, 2021-06-10:
Reopening this issue, since we have created a ticket on our backlog for this request.
Devs see CURA-8301

skorpi0n, 2021-06-10:
> This request is about adding tooltips to the icons in the header bar, not about other tooltips throughout the application.

Ghostkeeper is right about what I meant!