Ultimaker/Cura #9912: Unable to add Support Blockers

pjjoh1, 2021-05-27:
### Application Version

4.9.1

### Platform

Mac

### Printer

Creality Ender 3 v2

### Reproduction steps

1. After I select the model, 
2. press Support Blocker on the Left hand pane
3. click on the Model where I would like the support blocker to be

### Actual results

no box for the support blocker is created

Screen recording of behaviour can be found here: https://youtu.be/8YZQKchvYik

### Expected results

Support blocker box / object should appear on the build plate

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Screen capture of behaviour: https://youtu.be/8YZQKchvYik

[cura.log](https://github.com/Ultimaker/Cura/files/6553017/cura.log)

[CE3_Ender Camera C905 Bolts.3mf.zip](https://github.com/Ultimaker/Cura/files/6553024/CE3_Ender.Camera.C905.Bolts.3mf.zip)

fvrmr, 2021-05-27:
Hi @pjjoh1 thank you for your report. 
I can't reproduce it on Windows. But I will discuss this with the team. 
Keep you posted!

yet-another-average-joe, 2021-05-28:
This problem randomly happens to me from time to time...
(Windows 10)
The model posted by @pjjoh1 gives me this behaviour. After a few attempts (unselecting re-selecting, clicking...), I can place supports blockers.

pjjoh1, 2021-05-28:
Seemed to be able to fix this by deleting everything in the configuration folder and restarting Cura.

On Mac:
1. Help Menu -> Show Configuration Folder
2. Quit Cura
3. Delete everything in folder
4. Restart Cura

pjjoh1, 2021-05-29:
That only seemed to have worked until last restart.

pjjoh1, 2021-05-29:
This is more than likely linked to issue [9850](https://github.com/Ultimaker/Cura/issues/9850), where the model can't be selected.

pjjoh1, 2021-05-29:
Interestingly, this just doesn't work with Octoprint installed.  if I uninstall that plugin, I can select the model again **until** I go into Manager Printers.  After that time, the model can't be selected.

fvrmr, 2021-05-31:
Hi @pjjoh1 thank you for the updates, always really helpful. 

@yet-another-average-joe do you have the plugin Octoprint installed?

Our testers will look into this and see if they can reproduce this on Mac.

yet-another-average-joe, 2021-05-31:
Yes, the OctoPrint plugin is installed.

yet-another-average-joe, 2021-06-01:
Deactivating or uninstalling OctoPrint plugin makes no difference for me : the problem remains the same.

fvrmr, 2021-06-01:
Can you share your log file as well @yet-another-average-joe?

pjjoh1, 2021-06-02:
ok, progress.  Looks like it depends which monitor it's on.  Issue is really around not being able to select the model.  When Cura is on my Macbook's primary screen, everything works.  When Cura is on my secondary screen, I can't select the model by clicking on it, and hence can't place the blocker.

So simply putting cura on my primary model, I can effectively work around this problem.

This is working with and without octoprint installed and connected
Running a Macbook Pro 16" 2019, macOS Big Sur (11.4)

yet-another-average-joe, 2021-06-02:
> 
> 
> Can you share your log file as well @yet-another-average-joe?

@fvrmr : I now suspect the problem was on my side, not on Cura side : was always able to select the model itself. I just had random problems placing blockers, so I now think I was maybe misusing the feature (?)
BTW no problems on my Win10 quad monitor / quad GPU setup.

For now, please just ignore my posts.

evtrados, 2021-06-03:
I have also experienced this at times, but not constantly (at least on Catalina that I'm working on). Usually I make use of the keyboard shortcuts for stuff like this ('E' in the case of a support blocker) and it always seems to work.
So, I don't have reproduction steps for this issue. I tried multiple times to reproduce it this morning and I couldn't.

Ghostkeeper, 2021-06-07:
In the video (and something that's been said in this thread before), it kind of looks like the user has trouble selecting the model as well, meaning that something is going wrong with the rendering of the selection buffer. The selection buffer is an image that Cura draws in the background with a unique colour for every object. When clicking the scene, Cura will look in that image for the coordinates of the mouse where the click happened, and can then look up which 3D model was clicked on by looking at the colour of the pixel there.

If the selection buffer doesn't render at all, no clicks would have any effect. If it's too small or too big, you'd need to click next to the object (I'd expect to its bottom-left if the selection buffer is too small, or its top-right if the selection buffer is too big). If it's outdated, it would e.g. still click on the previous position of an object after scaling it.

My gut feeling says it's the second one. Perhaps it's rendering too big because it's rendering at the wrong screen scale. If you rotate the camera, the selection buffer would refresh and it would work again. Maybe that's why only some people see this issue?

jacix, 2021-06-15:
This problem started for me after version 4.8.0 (which still works perfectly) and I commented on it in another thread.  I hadn't even thought to try a different monitor (I have the laptop's built-in and 3 externals) but that seems to be the answer, at least short-term.   Cura-4.9* works fine on the internal Retina screen and the 27" LG UltraFine 5K - which is built for Macs.  On the two 27" QHD Dells, though, I can't select anything on the print bed, and mouse activity in general is erratic: rotation swivels around a seemingly random axis, zooming doesn't center where it should, etc.   Moving Cura to a different monitor shows the problem immediately in that it goes from showing the full printer bed to one corner.  I've attached screen-grabs: "happy-cura.png" is on the LG, and "unhappy-cura.png" is the same window on the far-right monitor.  Both captures were the full Cura window, so judging by the size of the image files, I'm guessing the application is having problems dealing with different resolutions.   As a test I changed the scaling on the LG and a Dell - it didn't make a difference: Cura works like a charm on the LG, but not on the Dell.  

In case it helps replicate the problem, here's more information about my setup.  All monitors have their resolution set to "Default for display".  From left to right:
* 27" Dell U2719DX
* 15" 2016 Macbook Pro - macOS Catalina 10.15.7 with the latest patches
* 27" LG UltraFine 5K
* 27" Dell U2719DX

If you need more detail, logs, etc, just let me know.

I can't even tell you how happy I am to have found this thread and to have Cura back!

[image]
[image]

jtusman, 2021-06-20:
I'm not sure why the thread keeps getting closed with no resolution.
The problem as reported in request #9111 is after installing Cura 4.9 I cannot click on any objects.
They cannot be moved or worked with at all. After completely removing  Cura and reinstalling it I can use it one time.
After shut down and restart it will not work again.
It was suggested to move the Cura to my MacBook screen. This does work.
After doing testing I found if I Keep my MacBook closed Cura 4.9.1  does work on my second monitor.

Ghostkeeper, 2021-06-21:
> I'm not sure why the thread keeps getting closed with no resolution.
> The problem as reported in request #9111 is after installing Cura 4.9 I cannot click on any objects.

This issue was never closed? The issue you link to is completely unrelated.

But indeed there is no resolution for this yet. We haven't been able to reproduce this issue yet, so far. All I can tell from the descriptions posted here is that some renders are likely being made at the wrong scale, so where you'd actually need to click is probably somewhere more towards the top-left or bottom-left of the screen. But how that could happen I don't know, and thus also not how to fix it.

jtusman, 2021-06-21:
When I drop a stl file on the build plate I cannot click on it and cannot
not or modify the object.
If I select it in the list at the bottom left of the screen it highlights
but cannot move or change it.
If I move Cura to my MacBook screen everything works as it should.
I also found that if I keep my MacBook closed Cura works on both of my
external monitors.
I hope this helps to understand what is happening.
Jack

jacix, 2021-06-22:
4.10-beta didn't help, though it also didn't make things worse. I swapped the 27" LG UltraFine 5K for its little brother, the 24" UtlraFine 4K, and the behavior is the same:  I can use Cura on the Macbook Pro's primary screen and the external (4K or 5K) LG, but not either of the 27" QHD (2K) Dells.

AmanC92, 2021-08-18:
I have the same issue as mentioned here, when cura is opened and opens on the primary laptop monitor and then the window is dragged to an external monitor the scaling gets messed up so you can no longer click on models. I have noticed this issue in previous application before where the window would not resize properly after changing monitor screens, and found a workaround so thought I'd post this here in case it wasn't mentioned before.

_Temporary Fix:_ Once you move the window from the laptop monitor to your external monitor, just manually resize the window in any direction by dragging the corners of the window. Once you resize your window the scaling is updated and you are able to select model again.

yepher, 2021-12-27:
This issue is super frustrating. Cura is almost unusable in it's current state

[image]

Attached is a `tail -f ~//Library/Application\ Support/cura/4.12/cura.log` from launch and while I try and select/move the object on the bed so I can print it:

I had no idea what the issue was until I found this thread. The problem is Cura launches on the last screen it was closed and even if you move it back to the laptop's built-in screen it does not work. You have to close Cura on the Laptop screen and then open it again to have any chance of manipulating the model on the Cura build platform.

[cura.log.zip](https://github.com/Ultimaker/Cura/files/7780875/cura.log.zip)

My machine is:
MacOS: 11.5.2
MacBook Pro (13-inch, 2016, Four Thunderbolt 3 Ports)
Graphics: Intel Iris Graphics 550 1536 MB
Cura 4.12

Displays
[image]

yepher, 2021-12-27:
I just read the comment from @AmanC92 and resizing the window makes think work again.

fieldOfView, 2021-12-27:
There's an option in the General preferences in Cura to restore the window position on start. Unchecking that will ensure Cura always opens on the primary display.

yepher, 2021-12-27:
Now that I know the workaround of `resizing the window` I am fine. The major problem was before I found this thread and spent hours trying to figure out how in the world to get various models oriented so I could print them. 

I hope it is an easy fix for the Cura team to figure out the root issue now that it is pretty clear that a simple window resize gets the app to being sane again.

Ghostkeeper, 2022-01-03:
With the next release we'll have to replace all QML code that generates windows with an update to Qt6.2. There's a chance that this fixes it.

0bsidian, 2022-02-24:
I have the same issue, but in my case, it does not work on my primary monitor and can only add support blockers on my secondary display.

Windows 10 Pro, 21H1
Cura 4.13.0

Procedure:

Select model, choose support blocker on the left side menu, can't create a support blocker by clicking on the model.

Repeating the same steps after moving Cura to my secondary display and sometimes it won't work, and other times it does work.

JJ-Atkinson, 2023-12-01:
FWIW I have this problem on cura 5.5, macos 14, m2 mac pro. I can't select or move my models when I have the program on my second (external) monitor.

closed 2024-11-19 by GregValiant