Ultimaker/Cura #9913: Serial port is opened on start, effectively blocking other software from using it

adzierzanowski, 2021-05-27:
### Application Version

4.9.1, 4.9.0, 3.6.0

### Platform

Windows 7

### Printer

Skriware 2

### Reproduction steps

1. Attach a device that uses FTDI USB-UART chip (commonly used in MKS boards)
2. Open Cura application
3. Try to open serial port for communication with the device

### Actual results

System denies access to the port:

`could not open port 'COM19': PermissionError(13, 'Permission denied', None, 5)`

When Cura is terminated, system allows access to the port.

### Expected results

User should be able to use the port while running Cura in the background.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I know that Windows 7 is old, ugly, bad and unsupported, yet sadly, it still makes up more than 16% of users as of 2021.

Our main concern is that the software we use for communication with some of our devices (which are not 3D printers) might be used with Cura opened. Our main target are schools where probability of Windows 7 installed on their PCs is quite high.

Our tests have shown that this problem doesn't occur on Windows 10 or macOS Big Sur.

fvrmr, 2021-05-27:
Hi @adzierzanowski thank your for your report.
Try disabling the USB plugin in the marketplace. (Go to marketplace - installed - disable USB printing).
Let me know if that works for you.

T-rex2017, 2021-06-07:
> 
> 
> Hi @adzierzanowski thank your for your report.
> Try disabling the USB plugin in the marketplace. (Go to marketplace - installed - disable USB printing).
> Let me know if that works for you.

is there a way to do this without opening the app

fieldOfView, 2021-06-07:
Find the configuration folder and open plugins.json in a text editor. Add "USBPrinting" to the "disabled" list, like so:
`{"disabled": ["USBPrinting"], "to_install": {}, "to_remove": []}`
Save the file and start Cura.

The configuration folder can be found here:
  * `%APPDATA%\cura\<Cura version>\` (Windows), or usually `C:\Users\\<your username>\AppData\Roaming\cura\<Cura version>\`
  * `$USER/Library/Application Support/cura/<Cura version>/` (OSX)
  * `$USER/.local/share/cura/<Cura version>/` (Ubuntu/Linux)

Ghostkeeper, 2021-06-07:
This is a duplicate of https://github.com/Ultimaker/Cura/issues/5207.

closed 2021-06-07 by Ghostkeeper