Ultimaker/Cura #9914: Move the irc channel

AiyionPrime, 2021-05-27:
### Is your feature request related to a problem?

**#cura** and **#cura-dev** are what's left over from my freenode journey.
Freenode recently made some drastic policy changes and should be avoided, *imho*.

### Describe the solution you'd like

I'd like to get rid of both freenode channels.
As the channel has been an annoyance ever since (#1277), we should consider moving to a different network.
Many, including freenodes staff, moved to https://libera.chat: their new platform.

I'd like cura to follow them there.

### Describe alternatives you've considered

Alternative networks would include [oftc](https://www.oftc.net/), [hackint](https://hackint.org/) and [ircnet](https://www.ircnet.com/).
I'd order them like above.

Another option would be to stay with freenode or start discussions about other platforms.

### Affected users and/or printers

There are not many users in the channels; but I found them pretty useful, when I tried to find an issue in the code a few years ago.

nallath, 2021-05-27:
I honestly even forgot that we still had those channels. I don't think i've even been there for 3 years now. 

Maybe @Ghostkeeper still has some info about this.

That being said, I have no issue with moving away from freenode (I'm much in favor of it actually, especially considering the current events).

AiyionPrime, 2021-05-27:
I haven't been active for ages; though I'm just now discussing an issue with someone called pingufan.

The user count in those channels is ridiculously low, though.

closed 2021-06-01 by nallath

Ghostkeeper, 2021-06-07:
I used to be "active" on those channels since I had IRC open in the background on boot. We'd get like 2 conversations per year. Very quiet.

However when I got a new work laptop I've forgotten to install it again, so I've not been online in the past 2 years either. Now that I work from home due to COVID-19, I'm not inclined to let it start on boot on my home laptop. But I might get it back on my work laptop.

Good to see that we moved to Libera :) I heard about the shenanigans going on at Freenode.

PR #9915 (2021-06-01): Update Readme regarding irc network
This would resolve #9914.
I think @Ghostkeeper documented the channels a few years ago, likely a good choice for a review.